balena-io/etcher #1627: Suggesstion: CLI could warn about unsupported file types

lurch, 2017-07-24:
- **Etcher version:** v1.1.0

In the GUI version of Etcher, if you have a file compressed with an unsupported compression scheme (e.g. `some_disk.img.7z`), then the file-selector won't allow you to select it (which I guess then prompts the user that they need to manually extract the uncompressed `some_disk.img` before proceeding).

However in the CLI version of Etcher you can pass a file such as `some_disk.img.7z` and it proceeds without any issues. A user might mistakenly believe that this is because the Etcher CLI supports (and decompresses) `.7z` files, but what _actually_ happens is that the Etcher CLI writes the still-compressed `some_disk.img.7z` to the drive.
It might be nice if the CLI could present a warning prompt (similar to the `This will erase the selected drive. Are you sure?` one) if it detects a file extension / mime-type that it doesn't natively support.

By contrast, for a compression scheme that Etcher *does* support (e.g. `some_disk.img.xz`), then Etcher **does** decompress the image and write the uncompressed data to disk.

jviotti, 2017-07-24:
Ah, we added labels at the same time :P

jviotti, 2017-07-24:
The change sounds great to me :+1:

closed 2018-10-15 by jviotti