balena-io/etcher #1629: Download links cause Google Chrome to alert about deception

ColdHeat, 2017-07-25:
Tried to download Etcher for OSX and received the following alert from Google Chrome. 
[image]

jhermsmeier, 2017-07-25:
Could you provide the exact version & link of the one that caused this to display?

lurch, 2017-07-25:
@jhermsmeier @jviotti @craig-mulligan
Yikes! I'm seeing exactly the same error (in Google Chrome Version 59.0.3071.11) for _any_ of the download links near the bottom of https://etcher.io/ :-(  Looks like Chrome is being a bit over-zealous?

jhermsmeier, 2017-07-25:
I'm also seeing it with Chrome Canary, but not in Chromium. I've reported it as false positive now, sadly not much more we can do about it, I think. And I can't help but wonder who reported GitHub's release assets as phishing...

imrehg, 2017-07-25:
Checking the domain that triggers the warning: https://www.google.com/transparencyreport/safebrowsing/diagnostic/#url=https://github-production-release-asset-2e65be.s3.amazonaws.com

imrehg, 2017-07-25:
Should be resolved now.

closed 2017-07-25 by lurch