balena-io/etcher #1632: Etcher CLI Accept Drive letter as drive info

codingyourlife, 2017-07-25:
It works if I write "1" instead of "D". But I wan't to call the CLI via C# via powershell and I am not aware of any way to get the drive number, just the letter.
- **Etcher version:** 1.0.0
- **Operating system and architecture:** Win10
- **Error** We can't find D on your system. Did you unplug the drive?

Command: `./etcher index.js --no-check 2017-07-05-raspbian-jessie.zip --drive D --yes --unmount`

jviotti, 2017-07-25:
Hi @codingyourlife,

In Windows, drive letters are "disk volumes", which are basically partitions. When flashing an image, you want to flash to the whole drive, not to a partition, so Etcher takes a "physical drive" instead, which is something like `\\.\\PHYSICALDRIVE1`. We basically grab the last letter from that string, so passing `1` happens to work :P 

I'm not on Windows at the moment, but WMI can provide you with the volume -> physical drive mapping. For example, try running `wmic diskdrive list brief`. 

I guess this is particularly confusing because on the GUI, we show the list of drive letters instead of the physical drive address for convenience. 

In any case, I think we can modify the CLI to take a drive letter, and do the mapping internally.

lurch, 2017-07-26:
@codingyourlife You might also want to look at https://github.com/resin-io-modules/drivelist

closed 2018-10-15 by jviotti