balena-io/etcher #1634: Debian repo broken?

grash54, 2017-07-26:
- **Etcher version:1.1.0**
- **Operating system and architecture:Debian Sid amd64**
- **Image flashed: None**
- **Do you see any meaningful error information in the DevTools? No**

When trying to update receiving the following message:
W: Skipping acquire of configured file 'etcher/binary-amd64/Packages' as repository 'https://dl.bintray.com/resin-io/debian stable InRelease' does not seem to provide it (sources.list entry misspelt?)

lurch, 2017-07-26:
Yeah, already looking into this (I spotted it myself) and have filed a support ticket with Bintray ;-)

Enigma0, 2017-07-27:
Same issue here.

```
E: Failed to fetch https://dl.bintray.com/resin-io/debian/dists/stable/InRelease  Unable to find expected entry 'etcher/binary-amd64/Packages' in Release file (Wrong sources.list entry or malformed file)
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

lurch, 2017-07-27:
This is now fixed. Apologies for the inconvenience.

closed 2017-07-27 by lurch

GreboZero, 2018-04-13:
I am also failing to receive keys when trying to add:

sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61

lurch, 2018-04-14:
@GreboZero See e.g. #1786 
It should work if you try again later...