balena-io/etcher #1638: Error when updating source list on Debian Stretch

fabiomoc, 2017-07-26:
- **Etcher version: 1.1.0**
- **Operating system and architecture: Debian Stretch**

This is the error I get when a I execute the command **apt update**:

E: Failed to fetch https://dl.bintray.com/resin-io/debian/dists/stable/InRelease  Unable to find expected entry 'etcher/binary-amd64/Packages' in Release file (Wrong sources.list entry or malformed file)
E: Some index files failed to download. They have been ignored, or old ones used instead.

The source list is in this path: /etc/apt/sources.list.d/etcher.list
And it's content is: **deb https://dl.bintray.com/resin-io/debian stable etcher**

Fred-Barclay, 2017-07-27:
Duplicate of #1634 😄

lurch, 2017-07-27:
Closing as duplicate

closed 2017-07-27 by lurch

fabiomoc, 2017-07-27:
Sorry. Thanks anyway, guys.

sherminator666, 2018-03-26:
Failed to fetch https://dl.bintray.com/resin-io/debian/dists/stable/InRelease 
pls solve the issue

jhermsmeier, 2018-03-26:
@sherminator666 see https://github.com/resin-io/etcher/issues/1634 – it's supposed to be https://dl.bintray.com/resin-io/debian/dists/stable/Release