balena-io/etcher #1639: "Error: Couldn't scan the drives: An unknown error occurred" when launching Etcher (Windows)

WasabiFan, 2017-07-27:
- **Etcher version:** 1.1.1
- **Operating system and architecture:** Windows 10 x64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?**

```
{"stack":"Error: Couldn't scan the drives: An unknown error occurred","message":"Couldn't scan the drives: An unknown error occurred"}
console.(anonymous function) @ C:\Users\<user>\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51
```

This error appears immediately after launching the app. It leaves Etcher in a state where I can go through the normal image selection process, but the "select a drive" step is never enabled.

lurch, 2017-07-27:
Duplicate of https://github.com/resin-io-modules/drivelist/issues/198

jhermsmeier, 2017-07-27:
@lurch technically yes, but given this is also an issue within Etcher, let's tie this to #1637 and close this once that's through.

closed 2017-08-02 by jviotti

jviotti, 2017-08-03:
Can you try the following snapshot build that contains the fix: https://drive.google.com/file/d/0B7tkbonGU-RyTUNKaVpLb2VCZ1E/view?usp=sharing? If that still doesn't work, can you set the `DRIVELIST_DEBUG` environment variable and execute the application from `cmd.exe`, so you can see the debug logs?

For example:

```
set DRIVELIST_DEBUG=1
cd C:\Program Files\Etcher
etcher.exe
```

wakitcha, 2017-08-04:
I ran into the same issue and downloaded the snapshot build.  It exhibits the same error.  Here are my debug logs:

<details>
[drivelist] Initializing scanner

[drivelist] Initializing COM
[drivelist] Initializing COM security levels
[drivelist] Creating connection instance
[drivelist] Connecting to server
[drivelist] Setting proxy security levels
[drivelist] Interpreting error 0x0
[drivelist] Getting list of drives from WMI
[drivelist] Getting id
[drivelist] Processing \\.\PHYSICALDRIVE1
[drivelist] Getting caption
[drivelist] Getting disk information
[drivelist] Opening handle on \\.\PHYSICALDRIVE1
[drivelist] Getting drive geometry
[drivelist] Getting id
[drivelist] Processing \\.\PHYSICALDRIVE0
[drivelist] Getting caption
[drivelist] Getting disk information
[drivelist] Opening handle on \\.\PHYSICALDRIVE0
[drivelist] Getting drive geometry
[drivelist] Getting system volume drive letter
[drivelist] Getting available volumes
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Getting volume type
[drivelist] Ignoring, drive type not removable nor local
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Interpreting error 0x80070006
</details>

lurch, 2017-08-05:
@jviotti I managed to find that error code at https://msdn.microsoft.com/en-us/library/windows/desktop/dd542643(v=vs.85).aspx
**E_HANDLE** `0x80070006` Invalid handle

@wakitcha Could you tell us what drives you have connected to your computer (both internal and external), in case that helps us debug the problem?

jviotti, 2017-08-07:
@lurch We are the ones returning that error. I sent https://github.com/resin-io-modules/drivelist/pull/204, which gets the actual error message, so we should catch it next time. I plan to release tomorrow, so hopefully we'll have enough data to fix the error very soon.

jviotti, 2017-08-07:
@wakitcha Oh, could it be happening because of inaccessible network drives? I'll test that soon.

wakitcha, 2017-08-07:
I did some additional testing this morning based on the disconnected network drive hypothesis.  I deleted all of the inaccessible drives and was able to run Etcher successfully in some cases.  Specifically, if Etcher runs as Administrator it worked.  Etcher did not request admin rights by default.  I'm guessing that's just this debug build?

If not run as administrator, I get this log.  With this I had four drives.  C:, D:, H: (mapped to a folder in D:), and E: being the SD card.

[drivelist] Initializing scanner
[drivelist] Initializing COM
[drivelist] Initializing COM security levels
[drivelist] Creating connection instance
[drivelist] Connecting to server
[drivelist] Setting proxy security levels
[drivelist] Interpreting error 0x0
[drivelist] Getting list of drives from WMI
[drivelist] Getting id
[drivelist] Processing \\.\PHYSICALDRIVE1
[drivelist] Getting caption
[drivelist] Getting disk information
[drivelist] Opening handle on \\.\PHYSICALDRIVE1
[drivelist] Getting drive geometry
[drivelist] Getting id
[drivelist] Processing \\.\PHYSICALDRIVE0
[drivelist] Getting caption
[drivelist] Getting disk information
[drivelist] Opening handle on \\.\PHYSICALDRIVE0
[drivelist] Getting drive geometry
[drivelist] Getting system volume drive letter
[drivelist] Getting available volumes
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Interpreting error 0x80070006

wilson735, 2017-08-07:
If it helps any, I'm getting this error when I attempt to run Etcher as an Administrator.

```
Couldn't scan the drives: An unknown error occurred
```

Followed by:
[image]

I currently have 2 local drives, 1 "USB Drive" (actually an adapter holding the microSD card that I wish to flash to), 1 RamDrive, and 1 network drive.

lurch, 2017-08-10:
@mgalde Thanks for your report. Can you confirm that you're using the latest version 1.1.2 of Etcher?

mgalde, 2017-08-10:
Sorry, yes the newest version was tried and also had the same issue. As I was trying to diagnose this issue I tried the version listed above and the 32bit version on download page. Once I unlocked my drives I was able to get it to run with no issue on the current 64bit version.

lurch, 2017-08-10:
@mgalde This particular issue (1639) has been closed, as the particular problem it was addressing has been fixed. It sounds like your problem is slightly different, so would you mind opening a new issue for it please?

mubed, 2017-08-29:
The problem isn't solved yet. I yesterday downloaded the newest version (1.1.2) and have this problem (Win7):
[image]

clperez, 2017-08-30:
HI. I got the same error on etcher 1.1.2 as well. 
Windows 10 in my case, 64 bits.

mattzuba, 2017-09-06:
I had a similar and possibly related error and narrowed it down to a 'bind-mounted' folder as a drive.  I have a folder in my documents folder that is mounted as my A: drive and Etcher would throw the error about permission denied for scanning drives.  Once I unmounted the drive, Etcher worked.

serdasteclas, 2017-09-10:
I've just performed a fresh windows install and installed the latest etcher, I noticed the problem only occurs when my bitlocker drive is locked, after unlocking the drive etcher will work as expected.

rakete, 2017-09-17:
@serdasteclas is right. I can verify this problem occurs when there is a locked Bitlocker drive present. It goes away once I unlock it.

gpresland, 2017-09-22:
Same issue.
Windows 10, x64.

Error only occurs if my BitLocker drive is unlocked. Problem goes away if I unlock my BitLocker drive.

behel33, 2017-10-01:
I had the same problem.
Etcher is in conflit with virtual drive "L:" on my system Windows 10.
To avoid it, I remove my virtual drive (ex subst \d L: )