balena-io/etcher #1643: Chrome OS version

Jop-V, 2017-07-28:
Chrome OS is currently lacking a graphical app that allows you to write an ISO to a USB stick or SD card. The only way to do it is to enable developer mode and use `dd`.

As Etcher uses web technologies it could become the first image writer for Chrome OS. I don't know how much effort it is to adapt from Electron/Node.JS to the Chrome application API.

jviotti, 2017-07-30:
We explored this at some point, but the Chrome API doesn't provide us the type of low level functions we need to implement something like Etcher. 

I'm aware the Etcher runs fine on ChromeOS under developer mode though (with a DE like XFCE)

closed 2017-07-30 by jviotti

dragon788, 2018-11-14:
The Chrome OS Recovery app can write an arbitrary image to USB, though I hadn't previously tested whether that includes ISOs. Today I had some time and a morbid curiosity so I confirmed that it DOES WORK. The recovery tool is fairly "dumb", so if you make the extension the one it recognizes (ie `.bin` ) then it will let you select the file to write to a SD card/USB drive. **BE CAREFUL** with this as it may trash your USB drive if you write a random non-image file to it.

- Download your ISO.
- Open the Files app and navigate to Downloads, then right click (two finger click) and "Rename" the file from `ubuntu-18.04.1-desktop-amd64.iso` to `ubuntu-18.04.1-desktop-amd64.bin` (which is just a 3 letter edit) or `anything-you-want.bin` if you are feeling bored/silly.
- Install the [Chrome Recovery App from the Chrome Webstore](https://chrome.google.com/webstore/detail/chromebook-recovery-utili/jndclpdbaamdhonoechobihbbiimdgai). 
- Open the Chrome Recovery App (named Recovery with the Chrome icon with a :wrench: over it in the launcher)
- The first time you open the app, the gear icon :gear: may not immediately show up, if that is the case then hit the "Get started" button to go to the next page and it should show up.
- Click the gear :gear: at the top right, and then "Use local image".
- Navigate to Downloads or wherever your ISO with the `.bin` extension is located and select it then click "Open".
- Choose the target device from the list, SD cards or USB drives are valid, it recommends a 4GB+, not sure if that is hardcoded but it is a good recommendation.
- Writing the ISO/image will take quite a while, but it should succeed. 

The tool can also do a bin from inside a zip file, but trying to fake compressing an ISO (they don't compress well if their creators were smart) after changing the extension to `.bin` is silly. It also requires way more space because it needs to extract the image and then write it (it might stream it, but the linux-recovery.sh alternative script isn't that smart for sure).

When you are finished with a recovery drive you can also reset it back to a normal drive from the gear :gear: menu using "Erase recovery media".

lurch, 2019-02-24:
I don't have a Chromebook myself, but [this PR](https://github.com/raspberrypi/documentation/pull/1090) also confirms that you can use "Chromebook Recovery Utility" to flash regular image files to an SD card.
@thundron I dunno if this is something worth adding to the Etcher FAQ?

user951, 2020-02-06:
true