balena-io/etcher #1644: I can't burn the Porteus ISO on an external HD

gusbemacbe, 2017-07-29:
- **Etcher version:** 1.1.1
- **Operating system and architecture:** macOS Sierra 10.12.6 64bits
- **Image flashed:** Porteus.iso
- **Do you see any meaningful error information in the DevTools?** No

 Nothing

Etcher doesn't detect the external HDs to burn the ISO of Porteus. I do not want to burn the ISO of Porteus on a SD card or USB driver for being very small than an external HD, because I want to run a portable OS on any computer. 

Here is Porteus site: http://www.porteus.org/

jhermsmeier, 2017-07-29:
@gusbemacbe have you tried enabling unsafe mode in the settings?

lurch, 2017-07-31:
From reading http://www.porteus.org/component/content/article/26-tutorials/general-info-tutorials/114-official-porteus-installation-guide.html#Hard%20Drive and http://www.porteus.org/component/content/article/26-tutorials/general-info-tutorials/114-official-porteus-installation-guide.html#USB-linux it sounds like even if you **did** use Etcher to write the Porteus ISO to your USB HD, it still wouldn't work anyway, as it involves running custom extra steps. Seems like the Porteus ISO is a 'real' ISO and not a '[hybrid ISO](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos)' like Ubuntu uses, and thus Porteus can't be extracted directly to a USB device.

As @jhermsmeier has already said, Etcher doesn't detect HardDrives by default (to prevent them being accidentally overwritten), unless you enable Etcher's "unsafe mode".

I'm closing this issue for now, but please add additional comments if you're still having problems and we'll re-open it.

closed 2017-07-31 by lurch