balena-io/etcher #1646: Couldn't initialize the scanner

stevie86, 2017-07-30:
- **Etcher version:** 1.1.1
- **Operating system and architecture:** Windows 10 x64
- **Image flashed:** none, not working
- **Do you see any meaningful error information in the DevTools?**
How do I do that on Windows
[image]

jviotti, 2017-07-30:
Related to https://github.com/resin-io-modules/drivelist/issues/198. Do you have a virtual drive mounted? Can you confirm that unmounting the virtual drive makes it work?

lurch, 2017-07-31:
See also #1639

MetalRush, 2017-08-01:
@ jviotti; I can confirm that (in my case) my Alcohol 120% virtual drive was the culprit.
I disabled the drive and reinstalled Etcher.
(now I have to find out why my image seems corrupt when validating)

closed 2017-08-02 by jviotti

jviotti, 2017-08-03:
Can you try the following snapshot build that contains the fix: https://drive.google.com/file/d/0B7tkbonGU-RyTUNKaVpLb2VCZ1E/view?usp=sharing? If that still doesn't work, can you set the `DRIVELIST_DEBUG` environment variable and execute the application from `cmd.exe`, so you can see the debug logs?

For example:

```
set DRIVELIST_DEBUG=1
cd C:\Program Files\Etcher
etcher.exe
```

mikaeljonssonhbg, 2017-08-05:
Hi,

Had the same issue and snapshot build from jviotti with fix above solved it for me! Thankxxx!

stevie86, 2017-08-21:
Hi! Somehow it works for me now, I don't know why! Thanks!

lurch, 2017-08-21:
@stevie86 Maybe you upgraded from Etcher 1.1.1 to Etcher 1.1.2  :-)

jviotti, 2017-08-21:
@stevie86 Did you have a virtual volume or some other kind of "special" volume that you unmounted?