balena-io/etcher #1656: Could not initialize scanner during the opening...

fredericklussier, 2017-08-02:
- **Etcher version: 1.1.1**
- **Operating system and architecture: Windows 10**
- **Image flashed: **
- **Do you see any meaningful error information in the DevTools?**

I was so happy before this update.

Got the error during the opening.
Than I select the file to use to flash
Press the select disk, no disk to select.

Retry it with version 1.0.0, nothing works. 

After one day of trying different think, I have to reformat the disk and reinstall everythink. With version 1.0.0 it works, don't ask me to di your test with version 1.1.1.

jviotti, 2017-08-02:
Duplicated of https://github.com/resin-io/etcher/issues/1646, which was recently fixed. Can you confirm that you had virtual drives?

closed 2017-08-02 by jviotti

jhermsmeier, 2017-08-03:
@jviotti this also happens with spun down HDDs, while they're spinning back up again (that's what I encountered)

wisteso, 2017-08-03:
@jviotti this is happening for me without any virtual drive - I also dont think it has to do with a spun down drive.

odd thing is that when I originally opened it, I kept it open for a good 30 min without doing anything. I left my desk (with it still being idle) and came back to the error message.

jviotti, 2017-08-03:
Can you try the following snapshot build that contains the fix: https://drive.google.com/file/d/0B7tkbonGU-RyTUNKaVpLb2VCZ1E/view?usp=sharing? If that still doesn't work, can you set the `DRIVELIST_DEBUG` environment variable and execute the application from `cmd.exe`, so you can see the debug logs?

For example:

```
set DRIVELIST_DEBUG=1
cd C:\Program Files\Etcher
etcher.exe
```

wisteso, 2017-08-03:
Yes, that build seems to work.

Two notes though : 
1) I was using the portable version of 1.1.1 and your fix was an installer. Don't think it matters, but just FYI. 
2) I noticed the error didnt occur with 1.1.1 if my SD card wasn't inserted. Only when it was inserted.

jviotti, 2017-08-03:
Thanks a lot for confirming the fix @wisteso. I should have uploaded both portable and installer snapshot builds, my bad!