balena-io/etcher #1659: [Feature Request] Custom Install Directory

Modelmat, 2017-08-03:
Is it possible to add a customisable install location?

Thanks

lurch, 2017-08-03:
On which platform? ;-)

jviotti, 2017-08-03:
I guess that should be Windows (we don't have "installers" in other platforms). This means we would get rid of the sweet one-click installer, but I don't have strong opinions on that. What do you think @lurch @jhermsmeier @Shou ?

lurch, 2017-08-03:
I don't tend to use Windows much, but given the preference I think more control is better than less (but then I also consider myself an 'advanced' user).

jhermsmeier, 2017-08-04:
I agree, it'd be nice to have the option.

Modelmat, 2017-08-04:
I am talking about the Windows Platform

tcrory, 2017-10-28:
At the very least, it would be nice if you could choose between installing for one user (seems to be the current default setting), and installing for all users under either C:\Program Files or C:\Program Files (x86) as appropriate.

lurch, 2017-10-28:
BTW I noticed the other day that if you have Etcher v1.0.0 installed on Windows, and then install Etcher v1.1.2, you then get two separate "Etcher" shortcuts on your desktop!  (IIRC one points to ProgramFiles, and the other points to AppData)

lurch, 2018-02-26:
I guess using the Windows Portable version, and then moving it to a directory of your choice, might be a workaround :wink:

craigusus, 2018-05-04:
+1 on the option of where to install. Not everyone wants programs installed to \AppData\Local\Programs\. Plus EACH user of that PC will have to download and install it separately. Please add the option to install for the 'Current User' or 'All Users'. It's only one more click.

Also noticed that v1.1.1 was in Program Files (x86) now 1.4.4 is in AppData\Local\Programs. So I also had 2 versions installed.

closed 2018-10-15 by jviotti