balena-io/etcher #1660: Couldn't Initalise The Scanner: An Unknow Error Occurred

Modelmat, 2017-08-03:
- **1.1.1**
- **Win 10 Edu x64**
- **No meaningful error information in the DevTools**

Just this and then cannot find USB

jhermsmeier, 2017-08-03:
Duplicate of #1646

jhermsmeier, 2017-08-03:
Do you have any virtual drives, virtual file systems or spinning HDDs attached to your system?

closed 2017-08-03 by jhermsmeier

Modelmat, 2017-08-03:
I do not have any virtual drives, VFS and I do have a secondary HDD.

jviotti, 2017-08-03:
Can you try the following snapshot build that contains the fix: https://drive.google.com/file/d/0B7tkbonGU-RyTUNKaVpLb2VCZ1E/view?usp=sharing? If that still doesn't work, can you set the `DRIVELIST_DEBUG` environment variable and execute the application from `cmd.exe`, so you can see the debug logs?

For example:

```
set DRIVELIST_DEBUG=1
cd C:\Program Files\Etcher
etcher.exe
```