balena-io/etcher #1662: v1.1.1 does not install on this Windows 10 running Surface Pro 4. It popes up a dialog about being unable to scan drives.

ZiCog, 2017-08-03:
- **Etcher version:**
1.1.1
- **Operating system and architecture:**
Windows 10 on Surface Pro 4
- **Image flashed:**
None.
- **Do you see any meaningful error information in the DevTools?**
Fails on install with an dialog box indicating some error that it cannot scan drives.

jhermsmeier, 2017-08-03:
Duplicate of #1646

closed 2017-08-03 by jhermsmeier

jviotti, 2017-08-03:
Can you try the following snapshot build that contains the fix: https://drive.google.com/file/d/0B7tkbonGU-RyTUNKaVpLb2VCZ1E/view?usp=sharing? If that still doesn't work, can you set the `DRIVELIST_DEBUG` environment variable and execute the application from `cmd.exe`, so you can see the debug logs?

For example:

```
set DRIVELIST_DEBUG=1
cd C:\Program Files\Etcher
etcher.exe
```