balena-io/etcher #1665: Couldn't scan the drives: An unknown error occurred

maxgrass, 2017-08-03:
- **Etcher version: 1.1.1**
- **Operating system and architecture: Win10 x64**
- **Image flashed: --**
- **Do you see any meaningful error information in the DevTools?**

[image]

[image]

[image]

{"stack":"Error: Couldn't scan the drives: An unknown error occurred","message":"Couldn't scan the drives: An unknown error occurred"}

Everything works with version 1.0.0

Tried with both "-portable" and installer version.

lurch, 2017-08-03:
Duplicate of #1646

closed 2017-08-03 by lurch

jviotti, 2017-08-03:
@lurch Do you think we should post links to the snapshot builds here?

lurch, 2017-08-03:
Good call :)   (I wasn't sure how soon you were planning on releasing 1.1.2, given that this bug seems to be affecting multiple people?)

jviotti, 2017-08-03:
I was planning to release today/tomorrow, but I rather do it on Monday, to avoid having to spend my weekend hot fixing stuff :P

jviotti, 2017-08-03:
Can you try the following snapshot build that contains the fix: https://drive.google.com/file/d/0B7tkbonGU-RyTUNKaVpLb2VCZ1E/view?usp=sharing? If that still doesn't work, can you set the `DRIVELIST_DEBUG` environment variable and execute the application from `cmd.exe`, so you can see the debug logs?

For example:

```
set DRIVELIST_DEBUG=1
cd C:\Program Files\Etcher
etcher.exe
```

maxgrass, 2017-08-03:
It's working. Thank you.

lurch, 2017-08-03:
@jviotti Out of curiosity, how come you're linking to a snapshot build on google drive rather than one of the snapshot builds in our S3 bucket?

jviotti, 2017-08-03:
@maxgrass Thanks a lot for confirming this!

@lurch I accidentally broke snapshot builds in a previous commit. See https://github.com/resin-io/etcher/pull/1666

mwsasser, 2017-08-04:
The snapshot build is working for me as well, the release version was not.

hadjilucasL, 2017-08-05:
I suspect in my case it was failing to scan because I had a floppy disk drive attached to the system. The floppy drive appears as a disk path but it will error out if there is no disk in the drive. 

```
C:\>A:
The device is not ready.

C:\>
```

The snapshot build works but appears to continuously poll the drive for access until a drive is finally selected. 

Windows 10 x64, Floppy Drive at (A:)

lurch, 2017-08-05:
> I had a floppy disk drive attached to the system

Wow, old-skool :sunglasses: :wink: 

> appears to continuously poll the drive for access until a drive is finally selected.

Yeah, Etcher re-runs `drivelist` every two seconds, to take account for USB drives / SD cards / etc. being inserted and removed (unfortunately repeated polling is the only reliable cross-platform way to get this info). @jviotti @jhermsmeier I wonder if we added a user-configurable "delay betweeen drivelist-rescans" that'd make the Etcher experience less bad for @hadjilucasL ?

jviotti, 2017-08-07:
@lurch Recent changes cause drivelist to only poll X amount of seconds **after** the previous run completed, so that should be enough to mitigate these kinds of issues.

drencorxeen, 2017-08-07:
@maxgrass,  what you posted does work as a work around.

My question is shouldn't the application be able to run without setting this value in a command prompt?

@lurch ,

Wouldn't that be old school :P :)

mwsasser, 2017-08-07:
I wonder if the release version didn't like my WevDAV network mapped drives?

jviotti, 2017-08-07:
@drencorxeen 

> My question is shouldn't the application be able to run without setting this value in a command prompt?

The environment value I suggested above is just to get more debugging information in case the snapshot build didn't work.

jviotti, 2017-08-07:
@mwsasser Could be! The fix seems general enough to cover most cases. We basically check if we are not allowed to get the device number of a certain drive, and if so omit it.

jviotti, 2017-08-08:
Etcher v1.1.2 is available at https://github.com/resin-io/etcher/releases/tag/v1.1.2. Let me know how it goes!

mubed, 2017-08-29:
I still have this problem in Version 1.1.2 (Win7):
[image]