balena-io/etcher #1673: Error: Couldn't scan the drives: An unknown error occurred

drencorxeen, 2017-08-05:
- **Etcher version:**
1.1.1
- **Operating system and architecture:**
Windows 10 64bit version 1703
- **Image flashed:**
Never got to the point to open a image to use
- **Do you see any meaningful error information in the DevTools?**
Don't have any DevTools on my gaming system.
Only error message I have is from Etcher itself:
Top left of pop up box is:  Error
Main body of pop up box is:
Couldn't scan the drives: An unknown error occurred
Error: Couldn't scan the drives: An unknown error occurred

I tried Windows 64bit Etcher download, Windows 32bit Etcher download, and Windows Portal Etcher download.  All of the Windows versions give me the same error message.

I tried running the program both in normal without admin and launch as admin and didn't change anything.

Version 1.0.0 worked fine on my system.

drencorxeen, 2017-08-05:
Finally got the log option:

[-1501911979411_log.txt](https://github.com/resin-io/etcher/files/1202141/-1501911979411_log.txt)

lurch, 2017-08-05:
Please see https://github.com/resin-io/etcher/issues/1665#issuecomment-320034516 and let us know if that fixes it for you.

Closing (for now) as duplicate of #1646

closed 2017-08-05 by lurch

drencorxeen, 2017-08-07:
Yes that as a ok work around, but shouldn't the application be able to do this on its own?

lurch, 2017-08-07:
Yes, we're planning to release a new version of Etcher soon with this fix included. Thanks for confirming that the fix works for you :-)