balena-io/etcher #1674: No way of overriding "Too Small for Image" restriction

Sawtaytoes, 2017-08-07:
- **Etcher version:** 1.1.1
- **Operating system and architecture:** Windows 10 64-bit
- **Image flashed:** Raspberry Pi Zero W
- **Do you see any meaningful error information in the DevTools?** N/A

The image is 16.0GB whereas the new microSD card is 15.9GB.

When using `dd` in Linux, I could easily copy this drive image to a USB drive, and it would work fine. I know I only have 2GB of data on the drive so it would work just fine when the oversized image file is copied ot the slightly smaller microSD card.

Going into dev tools, I manually changed the drive.size to be something larger than 16.0GB:
```js
t = angular.element($0).scope()

t.modal.toggleDrive(t.drive).then((...d) => console.log(d))
// Sun Aug 06 2017 20:22:30 GMT-0500 (Central Daylight Time) Toggle drive ({"drive":{"device":"\\\\.\\PHYSICALDRIVE3","raw":"\\\\.\\PHYSICALDRIVE3","description":"Generic STORAGE DEVICE USB Device","size":15931539456,"system":false,"protected":false,"mountpoints":[{"path":"F:"}],"name":"F:"},"previouslySelected":false})

t.drive.size = 17000000000
t.modal.toggleDrive(t.drive).then((...d) => console.log(d))
// {"stack":"Error: The drive is not large enough\n
```

I received an error when executing `t.modal.toggleDrive(t.drive)` at the function `selectionState.toggleSetDrive(drive.device);` in `controllers/drive-selector.js`.

Eventually it goes to the action creator in `exports.setDrive` where it's dispatching an action on the store.

```js
exports.setDrive = (drive) => {
  store.dispatch({
    type: store.Actions.SELECT_DRIVE,
    data: drive
  });
};
```

I couldn't find the reducer that runs when this event is passed in, but even with the modified `drive` object, it still complains "The drive is not large enough", but this time in a Windows popup with a JS stacktrace:
```ini
[Window Title]
Error

[Main Instruction]
The drive is not large enough

[Content]
Error: The drive is not large enough
    at Object.exports.createError (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\lib\shared\errors.js:253:17)
    at storeReducer (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\lib\shared\store.js:311:22)
    at Object.dispatch (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\node_modules\redux\lib\createStore.js:179:22)
    at Object.exports.setDrive (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\lib\shared\models\selection-state.js:34:9)
    at Object.exports.toggleSetDrive (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\lib\shared\models\selection-state.js:54:13)
    at shouldChangeDriveSelectionState.then (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\lib\gui\components\drive-selector\controllers\drive-selector.js:119:24)
    at processQueue (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:16843:37)
    at C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:16887:27
    at Scope.$digest (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:17982:15)
    at C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:18211:26
    at completeOutstandingRequest (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:6274:10)
    at C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:6554:7
    at wrapped (C:\Users\Kevin\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:386:29)

[OK]
```

jviotti, 2017-08-07:
Hi @Sawtaytoes ,

I'm not sure if we should ever allow a user to override a size check that is there for good reason (e.g. there is no way Etcher nor the user can know if the remaining data that is not being flashed will cause problems later on).

You can always manually cut an image with `dd`, and then flash the result.

What do you think @Shou @lurch @jhermsmeier ?

lurch, 2017-08-07:
I agree that this should be closed as invalid, and Etcher shouldn't change behaviour.
No point in making Etcher 'more dangerous' for the 0.001% of users who want to deliberately write a "too big" image onto their SD card ;-)

closed 2017-08-08 by jviotti