balena-io/etcher #1678: etcher.io website lists all downloads as linux

sarukuku, 2017-08-08:
I know this is not the place for this issue but I couldn't find a Twitter handle for etcher.io so I decided to drop this here. Just close this issue immediately, this is just a heads up.

[image]

[image]

lurch, 2017-08-08:
Haha, whoops! Pinging @craig-mulligan

BTW the 'correct' place for this issue would be https://github.com/resin-io/etcher-homepage/issues

craig-mulligan, 2017-08-08:
Taking a look @lurch, it's probably because we changed the releases format.

lurch, 2017-08-08:
@craig-mulligan I'm assuming https://github.com/resin-io/etcher-homepage/pull/92 fixes this?

craig-mulligan, 2017-08-08:
Yip, @lurch.

closed 2017-08-09 by sarukuku