balena-io/etcher #1681: Etcher does not work on Bodhi Linux 4.2

michmanbiker, 2017-08-08:
- *1.1.2**
- **Bodhi Linux 4.2 64 Bit:**
- **Image flashed:**
- **Do you see any meaningful error information in the DevTools?**
Etcher asks me for my Sudo password and then does nothing.

michmanbiker, 2017-08-08:
Etcher asks me for my Sudo password and then does nothing.

lurch, 2017-08-09:
I'll take a look at this later, to see if I can reproduce.

michmanbiker, 2017-08-09:
something to do with syslinux 6.03?

jviotti, 2017-12-04:
@lurch Did you make any progress on this ticket?

lurch, 2017-12-05:
Sorry, I forgot. I'll try to take a look soon.

lurch, 2017-12-06:
I've now tested this in a VirtualBox VM using the Bodhi 4.2 LiveCD. I used the 64-bit Etcher v1.2.0 AppImage to flash a virtual USB disk, and it all went fine, I didn't encounter any problems at all.

So apologies for the very slow response @michmanbiker but I'll mark this as closed. If you're still having problems, then try installing the 64-bit .deb file (it seems like Bodhi is an Ubuntu spin-off), and if you still can't get it working then please feel free to add extra comments here and I'll re-open.

closed 2017-12-06 by lurch