balena-io/etcher #1682: Created partition not recognized in Windows version

facetime88, 2017-08-09:
- **Etcher version:** 1.1.2
- **Operating system and architecture:** Windows 10 Pro x64
- **Image flashed:** Benja OS https://benja.io
- **Do you see any meaningful error information in the DevTools?** No

I have tried many times flash benja to my micro sd sandisk class 10 16GB. The result as follow:

- There are 3 created partition and 1 unallocated space (benja used only 8GB, so the unallocated is the rest 8GB)
-  The first partition is recognized and shown in Windows.
- The rest partition is not recognized. Only the first partition has its driver letter.

- I open mini tool partition wizard and it shows like this:
[image]

- The second partition should be ext4 but its shows as "OTHER"
- I have submit this issue to Benja. They said there is no problem in Linux. They use the flashed sd card burned by linux burner and then plug the sd card into Windows, and it works fine, all 3 partitions are accessable by Windows with each has its own drive letter.
- I have tried 2 different computer. One on this one (Windows 10 Pro x64) and the second is on my laptop Windows 10 Home x86, same result.
- I have tried to use another brand new 8GB sandisk class 10 micro sd, still same result.

jviotti, 2017-08-09:
Hi @facetime88 ,

Etcher **does not** manipulate the partitions of any image, and it preserves the partition table that comes with the image you select. The result you see is from the partition table Benja OS ships with by default. There are some flashers, that since target Linux distros specifically, do create a new partition table and move files around, which is the reason other people using different tools might be getting a different result.

Did you try booting the image? Unless it requires special treatment, like Windows images (which I highly doubt), it should work just fine.

closed 2017-08-09 by jviotti

facetime88, 2017-08-09:
I put the flashed sd card to my raspberry pi and it works fine. The OS can start the default electron app in the BENJA-APP partition. So it means the raspberry doesn't have any problem accessing the BENJA-APP partition (which is FAT32 file system). But I do not know why I can not access the partition in Windows.

WebReflection, 2017-08-09:
Benja "_team_" here (just me). I have used etcher 1.1.2-x64, downloaded the iso, unzipped, and burned, with validation.

[image]

[image]

The folder BENJA-APP started showing even before the end of the validation and a message asking me to format the partition (Linux Ext4) appeared and I've ignored / canceled such message.

[image]

All good, I can confirm the benja iso is just fine and Etcher worked like a charm.

[image]

@facetime88 I'm afraid I've no idea what's wrong with your Windows 10 Pro :disappointed:

WebReflection, 2017-08-09:
P.S. @facetime88 my windows **never** understood the second partition ... it doesn't have to recognize it to make the third one available.

facetime88, 2017-08-09:
Wait...what?? Work like a charm even burn it in Windows? This is a really weird situation for me. So basically you have done the same steps I made, and result in success while I am not.

Really weird....:D I think I need some time for vacation and hope when back everything will give me different result.

jviotti, 2017-08-09:
Thanks for reaching out @WebReflection !

@facetime88 I can confirm that as far as I know, Windows won't detect ext4 partitions, unless special software/drivers are installed, so the warning should be good to ignore.