balena-io/etcher #1683: Wishlist: etcher-cli build for ARM platform (Raspberry Pi Raspbian OS)

Liontooth, 2017-08-09:
It would be useful to run etcher-cli from an RPi.

jviotti, 2017-08-09:
Hi @Liontooth ,

Thanks for reaching out. Etcher already builds fine on armhf, however we're having a very hard time making Travis CI build it for that arch (see https://github.com/resin-io/etcher/pull/1482).

Until the issue is sorted,  you can run the project from source as described in https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md

closed 2017-08-09 by jviotti

cyberb, 2018-03-24:
Why cannot you just have few arm boards with build agents on them?
I understand Travis is free, but how many build agents do you need to own and have something like drone.io on them either onsite or in the cloud like scaleway. I am doing both for syncloud.org and will never go back to Travis.

jhermsmeier, 2018-03-26:
@cyberb we do have ARM build servers, and will eventually get around to building & packaging for ARM, but there are some other, a little more involved things to sort out first unfortunately – mainly electron-builder and some dependencies not supporting ARM