balena-io/etcher #1684: SystemRescueCD 5.0.3 ISO problem

plittlefield, 2017-08-09:
When loading the SystemRescueCD ISO file, Etcher warns that the file is not bootable.

So, I have to use the [Manual Method](http://www.system-rescue-cd.org/Installing-SystemRescueCd-on-a-USB-stick/) for making a bootable USB drive.

Etcher Version: etcher-1.1.2-x86_64.AppImage
Operating System: Ubuntu 16.04.2

lurch, 2017-08-09:
Sounds like this issue is very similar to https://github.com/resin-io/etcher/issues/1644 i.e. the problem isn't  with Etcher, but with the image file being a 'real' ISO.
@jviotti Should we close?

plittlefield, 2017-08-10:
I will try that suggestion and report back.

Paul

jviotti, 2017-08-10:
Awesome. I'll close it, but please re-open if you find something else :)

closed 2017-08-10 by jviotti