balena-io/etcher #1688: Etcher Arm Architecure

okumedia, 2017-08-11:
How do I compile etcher for Arm device running Debian?

lurch, 2017-08-14:
It's something I've not tried myself for ages, but we've had several users reporting that the instructions in https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md (used to be in `docs/RUNNING_LOCALLY.md`) work for them.
Feel free to add further comments if you run into problems.

Also, we're eventually hoping to publish ARM builds of Etcher (suitable for running on Raspbian on a Raspberry Pi 2 or Raspberry Pi 3) ourselves.

closed 2017-08-14 by lurch

jviotti, 2017-08-14:
There were some tweaks we needed to do, that were added to the armhf branch we never merged (https://github.com/resin-io/etcher/pull/1482), so probably worth checking out that branch, and follow the contributing guide on top of that.