balena-io/etcher #1689: Failed to unmount after writing PiAware to SD Card

gregcotech, 2017-08-12:
- **Etcher version: 1.1.2
- **Operating system and architecture: Windows 7 Pro 32Bit
- **Image flashed: piaware-sd-3.5.0
- **Do you see any meaningful error information in the DevTools?: Function.prototype.apply.call
- ** SD Card: Sandisk Ultra microSDHC UHS-I Card 16GB Speed up to 48MB/s

Flashed PiAware onto SD card - Successful
Validated Install: Successful
Unmount: Failed!

lurch, 2017-08-14:
Could you please post the full DevTools console output? Thanks!

CACTANAKA, 2017-08-22:
Hello.

Maybe I facing same issue.
My environment as follows.

- Windows 10 Pro 64bit
- Etcher 1.1.12
- 2017-07-05-raspbian-jessie-lite.zip

And I attach log file as follows.

 document not ready yet, trying again in 500 milliseconds...
 Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.1.2, Type = nsis
 Sun Aug 20 2017 21:42:12 GMT+0900 (東京 (標準時)) Application start ({"packageType":"nsis","version":"1.1.2"})
 Sun Aug 20 2017 21:42:12 GMT+0900 (東京 (標準時)) Checking for updates ({"currentVersion":"1.1.2","releaseType":"PRODUCTION","updateSemverRange":"<2.0.0","includeUnstableChannel":false})
 Sun Aug 20 2017 21:42:14 GMT+0900 (東京 (標準時)) Update notification skipped ({"reason":"Latest version"})
 Sun Aug 20 2017 21:43:49 GMT+0900 (東京 (標準時)) Open image selector
 Sun Aug 20 2017 21:44:06 GMT+0900 (東京 (標準時)) Select image ({"path":"C:\\rpi_image\\2017-07-05-raspbian-jessie-lite.zip","size":{"original":1725629563,"final":{"estimation":false,"value":1725629563}},"extension":"img","archiveExtension":"zip","hasMBR":true,"hasGPT":false,"partitions":[{"type":12,"id":null,"name":null,"firstLBA":8192,"lastLBA":93597,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":94208,"lastLBA":3370370,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})
 Sun Aug 20 2017 21:44:28 GMT+0900 (東京 (標準時)) Flash ({"image":"C:\\rpi_image\\2017-07-05-raspbian-jessie-lite.zip","drive":{"device":"\\\\.\\PHYSICALDRIVE1","raw":"\\\\.\\PHYSICALDRIVE1","displayName":"D:","description":"Ricoh SD Disk Device","size":15986589696,"system":false,"protected":false,"mountpoints":[{"path":"D:"}]},"uuid":"1468b03e-dd93-4cb8-94ef-e94b9fa9df6e","unmountOnSuccess":true,"validateWriteOnSuccess":true})
 WRITER: Attempting to elevate

 WRITER: Running: C:\Users\au987\AppData\Local\Programs\etcher\Etcher.exe C:\Users\au987\AppData\Local\Programs\etcher\resources\app.asar\lib\child-writer\writer-proxy.js C:\Users\au987\AppData\Local\Programs\etcher\resources\app.asar C:\rpi_image\2017-07-05-raspbian-jessie-lite.zip --drive \\.\PHYSICALDRIVE1 --unmount --check

 Sun Aug 20 2017 21:44:46 GMT+0900 (東京 (標準時)) Progress (write): 0% at 4.83 MB/s (eta 356s)
 Sun Aug 20 2017 21:44:46 GMT+0900 (東京 (標準時)) Progress (write): 0% at 6.14 MB/s (eta 280s)
 Sun Aug 20 2017 21:44:46 GMT+0900 (東京 (標準時)) Progress (write): 0% at 6.45 MB/s (eta 266s)
 Sun Aug 20 2017 21:44:47 GMT+0900 (東京 (標準時)) Progress (write): 0% at 6.26 MB/s (eta 274s)
 Sun Aug 20 2017 21:44:48 GMT+0900 (東京 (標準時)) Progress (write): 1% at 6.79 MB/s (eta 251s)
 Sun Aug 20 2017 21:44:48 GMT+0900 (東京 (標準時)) Progress (write): 1% at 7.47 MB/s (eta 228s)
 Sun Aug 20 2017 21:44:49 GMT+0900 (東京 (標準時)) Progress (write): 1% at 7.04 MB/s (eta 242s)
 Sun Aug 20 2017 21:44:49 GMT+0900 (東京 (標準時)) Progress (write): 1% at 8 MB/s (eta 212s)
 Sun Aug 20 2017 21:44:50 GMT+0900 (東京 (標準時)) Progress (write): 2% at 8.38 MB/s (eta 201s)
 Sun Aug 20 2017 21:44:50 GMT+0900 (東京 (標準時)) Progress (write): 2% at 8.7 MB/s (eta 193s)
 Sun Aug 20 2017 21:44:51 GMT+0900 (東京 (標準時)) Progress (write): 2% at 9.12 MB/s (eta 184s)
 Sun Aug 20 2017 21:44:51 GMT+0900 (東京 (標準時)) Progress (write): 3% at 9.44 MB/s (eta 177s)
 Sun Aug 20 2017 21:44:52 GMT+0900 (東京 (標準時)) Progress (write): 3% at 9.43 MB/s (eta 176s)
 Sun Aug 20 2017 21:44:52 GMT+0900 (東京 (標準時)) Progress (write): 3% at 9.66 MB/s (eta 172s)
 Sun Aug 20 2017 21:44:53 GMT+0900 (東京 (標準時)) Progress (write): 4% at 9.99 MB/s (eta 165s)
 Sun Aug 20 2017 21:44:53 GMT+0900 (東京 (標準時)) Progress (write): 4% at 10.06 MB/s (eta 164s)
 Sun Aug 20 2017 21:44:54 GMT+0900 (東京 (標準時)) Progress (write): 5% at 10.18 MB/s (eta 161s)
 Sun Aug 20 2017 21:44:54 GMT+0900 (東京 (標準時)) Progress (write): 5% at 10.22 MB/s (eta 160s)
 Sun Aug 20 2017 21:44:55 GMT+0900 (東京 (標準時)) Progress (write): 5% at 10.32 MB/s (eta 158s)
 Sun Aug 20 2017 21:44:55 GMT+0900 (東京 (標準時)) Progress (write): 6% at 10.55 MB/s (eta 154s)
 Sun Aug 20 2017 21:44:56 GMT+0900 (東京 (標準時)) Progress (write): 6% at 10.49 MB/s (eta 154s)
 Sun Aug 20 2017 21:44:56 GMT+0900 (東京 (標準時)) Progress (write): 6% at 10.66 MB/s (eta 151s)
 Sun Aug 20 2017 21:44:57 GMT+0900 (東京 (標準時)) Progress (write): 7% at 10.82 MB/s (eta 148s)
 Sun Aug 20 2017 21:44:57 GMT+0900 (東京 (標準時)) Progress (write): 7% at 10.89 MB/s (eta 146s)
 Sun Aug 20 2017 21:44:58 GMT+0900 (東京 (標準時)) Progress (write): 7% at 10.98 MB/s (eta 145s)
 Sun Aug 20 2017 21:44:58 GMT+0900 (東京 (標準時)) Progress (write): 8% at 11.13 MB/s (eta 142s)
 Sun Aug 20 2017 21:44:59 GMT+0900 (東京 (標準時)) Progress (write): 8% at 11.17 MB/s (eta 141s)
 Sun Aug 20 2017 21:44:59 GMT+0900 (東京 (標準時)) Progress (write): 9% at 11.28 MB/s (eta 139s)
 Sun Aug 20 2017 21:45:00 GMT+0900 (東京 (標準時)) Progress (write): 9% at 11.32 MB/s (eta 138s)
 Sun Aug 20 2017 21:45:00 GMT+0900 (東京 (標準時)) Progress (write): 9% at 11.41 MB/s (eta 136s)
 Sun Aug 20 2017 21:45:01 GMT+0900 (東京 (標準時)) Progress (write): 10% at 11.51 MB/s (eta 134s)
 Sun Aug 20 2017 21:45:01 GMT+0900 (東京 (標準時)) Progress (write): 10% at 11.5 MB/s (eta 134s)
[... cut ...]
(anonymous) @ C:\Users\au987\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:18211
completeOutstandingRequest @ C:\Users\au987\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:6274
(anonymous) @ C:\Users\au987\AppData\Local\Programs\etcher\resources\app.asar\node_modules\angular\angular.js:6554
wrapped @ C:\Users\au987\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:386
 Sun Aug 20 2017 21:53:05 GMT+0900 (東京 (標準時)) Modal accepted ({"value":false})

jviotti, 2017-08-25:
Hi @gregcotech , @CACTANAKA ,

Can you try the following custom build: https://drive.google.com/open?id=0B7tkbonGU-RyVlY4U1c5Z0g3akU? It includes the PR I sent linked to this issue, which might resolve the problem.

Let me know if it works!

gregcotech, 2017-08-25:
Hi Juan,

Thanks very much for your reply. I'm currently on holidays out of the country for a little while and back next week. I just like to acknowledge that I tried a few different options and finally got the issues resolved. Many thanks for your help and assistance.

Kind regards,

Greg

https://github.com/resin-io/etcher/issues/1689#issuecomment-324972792

CACTANAKA, 2017-09-01:
Hi @jviotti ,

Thank you for reply!
I'll try to your custom build at tomorrow. Have a little time.

Best regards,

gregcotech, 2017-09-01:
Thank you.

I already got sorted. Not sure I'd you got my previous email.
Thanks again.

Greg

CACTANAKA, 2017-09-02:
Hi @jviotti ,

I tried your version. But failed again.
(maybe, your code is worked correctly. retry logic is worked 5 times.)

I attach new log follows.

Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
app.js:74  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.1.2+7b791d6, Type = nsis
Sun Sep 03 2017 01:40:35 GMT+0900 (東京 (標準時)) Application start ({"packageType":"nsis","version":"1.1.2+7b791d6"})
Sun Sep 03 2017 01:40:35 GMT+0900 (東京 (標準時)) Checking for updates ({"currentVersion":"1.1.2+7b791d6","stable":false,"updateSemverRange":"<2.0.0","includeUnstableChannel":true})
Sun Sep 03 2017 01:40:38 GMT+0900 (東京 (標準時)) Update notification skipped ({"reason":"Latest version"})
Sun Sep 03 2017 01:40:46 GMT+0900 (東京 (標準時)) Open image selector
Sun Sep 03 2017 01:40:48 GMT+0900 (東京 (標準時)) Select image ({"path":"C:\\rpi_image\\2017-07-05-raspbian-jessie-lite.zip","size":{"original":1725629563,"final":{"estimation":false,"value":1725629563}},"extension":"img","archiveExtension":"zip","hasMBR":true,"hasGPT":false,"partitions":[{"type":12,"id":null,"name":null,"firstLBA":8192,"lastLBA":93597,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":94208,"lastLBA":3370370,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})
Sun Sep 03 2017 01:40:51 GMT+0900 (東京 (標準時)) Flash ({"image":"C:\\rpi_image\\2017-07-05-raspbian-jessie-lite.zip","drive":{"size":15986589696,"adaptor":"standard","raw":"\\\\.\\PHYSICALDRIVE1","displayName":"D:","protected":false,"system":false,"device":"\\\\.\\PHYSICALDRIVE1","pending":false,"mountpoints":[{"path":"D:"}],"description":"Ricoh SD Disk Device"},"uuid":"4e2d94bd-9f55-435e-9078-4e27b64f0d54","unmountOnSuccess":true,"validateWriteOnSuccess":true})
\app.asar\lib\child-writer\index.js:183 WRITER: Re-spawning with elevation

\app.asar\lib\child-writer\index.js:161 [mountutils] \\.\PHYSICALDRIVE1
\app.asar\lib\child-writer\index.js:161 [mountutils] Checking whether drive is ejectable: C
\app.asar\lib\child-writer\index.js:161 [mountutils] The drive is fixed
\app.asar\lib\child-writer\index.js:161 [mountutils] Opening drive letter handle: C
\app.asar\lib\child-writer\index.js:161 [mountutils] Closing drive letter handle
\app.asar\lib\child-writer\index.js:161 [mountutils] Continuing with the next available letter
\app.asar\lib\child-writer\index.js:161 [mountutils] Checking whether drive is ejectable: D
\app.asar\lib\child-writer\index.js:161 [mountutils] The drive is removable
\app.asar\lib\child-writer\index.js:161 [mountutils] Opening drive letter handle: D
\app.asar\lib\child-writer\index.js:161 [mountutils] Closing drive letter handle
\app.asar\lib\child-writer\index.js:161 [mountutils] Drive letter device matches
\app.asar\lib\child-writer\index.js:161 [mountutils] Ejecting drive letter
\app.asar\lib\child-writer\index.js:161 [mountutils] Creating volume handle
\app.asar\lib\child-writer\index.js:161 [mountutils] Locking volume
\app.asar\lib\child-writer\index.js:161 [mountutils] Dismounting volume
\app.asar\lib\child-writer\index.js:161 [mountutils] Ejecting volume
\app.asar\lib\child-writer\index.js:161 [mountutils] Retrying ejection
\app.asar\lib\child-writer\index.js:161 [mountutils] Retrying ejection
\app.asar\lib\child-writer\index.js:161 [mountutils] Retrying ejection
\app.asar\lib\child-writer\index.js:161 [mountutils] Retrying ejection
\app.asar\lib\child-writer\index.js:161 [mountutils] Retrying ejection
\app.asar\lib\child-writer\index.js:161 [mountutils] Couldn
\app.asar\lib\child-writer\index.js:161 '
\app.asar\lib\child-writer\index.js:161 t eject volume
\app.asar\lib\child-writer\index.js:161 [mountutils] Couldn
\app.asar\lib\child-writer\index.js:161 '
\app.asar\lib\child-writer\index.js:161 t eject drive letter
\app.asar\lib\child-writer\index.js:161 [mountutils] Unmount complete
Sun Sep 03 2017 01:48:05 GMT+0900 (東京 (標準時)) Flash error ({"error":{"message":"Unmount failed","stack":"Error: Unmount failed"},"image":"C:\\rpi_image\\2017-07-05-raspbian-jessie-lite.zip","drive":{"size":15986589696,"adaptor":"standard","raw":"\\\\.\\PHYSICALDRIVE1","displayName":"D:","protected":false,"system":false,"device":"\\\\.\\PHYSICALDRIVE1","pending":false,"mountpoints":[{"path":"D:"}],"description":"Ricoh SD Disk Device"},"uuid":"4e2d94bd-9f55-435e-9078-4e27b64f0d54","unmountOnSuccess":true,"validateWriteOnSuccess":true})
Sun Sep 03 2017 01:48:05 GMT+0900 (東京 (標準時)) Open modal ({"template":"./components/warning-modal/templates/warning-modal.tpl.html"})
\app.asar\node_modules\resin-corvus\src\resin…:226 {"stack":"Error: Unmount failed","message":"Unmount failed"}
logException @ \app.asar\node_modules\resin-corvus\src\resin…:226
exports.report @ \app.asar\lib\gui\modules\exception-reporter.…:39
ImageWriterService.flash.then.catch @ \app.asar\lib\gui\pages\main\controllers\flas…:98
processQueue @ \app.asar\node_modules\angular\angular.js:16843
(anonymous) @ \app.asar\node_modules\angular\angular.js:16887
$digest @ \app.asar\node_modules\angular\angular.js:17982
(anonymous) @ \app.asar\node_modules\angular\angular.js:18211
completeOutstandingRequest @ \app.asar\node_modules\angular\angular.js:6274
(anonymous) @ \app.asar\node_modules\angular\angular.js:6554
Sun Sep 03 2017 01:58:46 GMT+0900 (東京 (標準時)) Modal accepted ({"value":false})

lurch, 2017-09-04:
@gregcotech 

> I just like to acknowledge that I tried a few different options and finally got the issues resolved.

That's good to hear. For the benefit of any other users that run into the same problem, do you remember _which_ particular 'option' got things working properly?

CACTANAKA, 2017-09-28:
Hi
I did bypassed by following setting.

☑ Anonymously Report errors and usage statistics to resin.io
□ Eject on success
□ Validate write on success
☑ Include unstable update channel
□ Unsafe mode [Dangerous]

Just Information.

jhermsmeier, 2018-03-09:
Duplicate of #1370

closed 2018-03-09 by jhermsmeier