balena-io/etcher #1690: Status hangs at "Starting"

SRinStL, 2017-08-12:
- **Etcher version:**1.1.2
- **Operating system and architecture:**Mac OSX
- **Image flashed:**Raspian
- **Do you see any meaningful error information in the DevTools?**No, after selecting the image and the drive is selected automatically, the status changes to "Starting" after clicking "Flash" then just sits there with no status change even leaving it overnight.

lurch, 2017-08-14:
What version of macOS are you using? Which version of Raspbian are you trying to write? What disk / device / card are you trying to write to? If writing to an SD card, what SD-writer are you using?
Even if you don't think it's useful, could you please copy'n'paste your full DevTools console output?

SRinStL, 2017-08-16:
OSX and Raspbian are both updated to the latest version. I'm using a 32GB class 10 card from Lexan. I was able to get it running successfully after restarting my system. 

Spencer

lurch, 2017-08-16:
Hmmm, I guess that means you're not able to reproduce the problem? In that case I dunno if there's much we can do... :-/

SRinStL, 2017-08-16:
The best recommendation to repeat would be to uninstall it, install the previous version, then perform an upgrade. I had the prior version installed and did a few burns of SD cards over a couple of weeks. At the time of the error I reported, I had just installed the upgrade after being notified it was available when I was getting ready to burn another card. 

Spencer

lurch, 2017-08-16:
I personally don't have a Mac to test with - can any of the rest of the Etcher team reproduce this? @jviotti @jhermsmeier @Shou

jviotti, 2017-08-21:
I can't reproduce, but heard of another user with a similar issue. 

@SRinStL The unmounting/elevating code hasn't changed at all for a couple of releases, so a regression in the current version is very unlikely the problem. Maybe macOS is just been crazy and getting stuck at unmounting the drive. I remember experiencing this myself with `diskutil` a couple of times (the tool never exits and drives don't get unmounted)

ZS5LT, 2017-09-24:
**Etcher version:**1.1.2
**Operating system and architecture:**Win7 Pro
**Image flashed:**Skylark
**Do you see any meaningful error information in the DevTools?
**No, after selecting the image and the drive is selected automatically, the status changes to "Starting" after clicking "Flash" then just sits there with no status change.

invariant, 2017-11-02:
I can confirm I just had the exact same issue (macOS 10.12, Etcher 1.1.2, flashing raspbian stretch)

SvenDowideit, 2017-11-07:
I have something like this too.
```
Etcher: 1.1.2
OS: Linux debian, using i3, running with sudo, as i don't have polkit
Image: resin 2.2.0-test..
```

But... if I wait long enough (about 2mins), `Starting...` changes to 54%, and then sits there.

running `iostat -dx 1` I can see that we're maxed out writing to the SD card for about 2mins, then the utilisation goes to 0, and stays there. about 15 seconds? later, the `Starting` button changes  (this time `51%`) , and we never progress again.

jhermsmeier, 2017-11-17:
Duplicate of #1340

jhermsmeier, 2017-11-17:
Fixed in https://github.com/resin-io/etcher/pull/1849

closed 2017-11-17 by jhermsmeier

itsderek23, 2020-02-11:
I'm running OSX 10.14.6 and was able to get running by granting full disk access to Etcher:

http://osxdaily.com/2018/10/09/fix-operation-not-permitted-terminal-error-macos/

SimonSomlai, 2021-11-24:
I was able to fix this by;

1) Opening package contents of balena etcher program. Then go into `/Contents/MacOs`
[image]
2) Dragging the `balenaEtcher.sh` script into my terminal and typing "sudo" in front of it before executing