balena-io/etcher #1691: Emit events if the flashing process gets stuck

jviotti, 2017-08-14:
The GUI application expects to be updated about the writer process state very often through IPC. Given how many people get stuck at "Starting..." or at the middle of the flash, lets trigger a timeout every time the flash process start or every time we get an update from the writer process, and emit an informational event to Mixpanel if the timeout expires and we didn't get another update.

lurch, 2017-08-14:
I guess this also links (indirectly) to @Shou 's task about adding an 'Elevating...' state?

And I dunno if it relates directly to this issue or not, but I noticed that sometimes (when writing a smaller image) in the DevTools console the flashing progress might start at e.g. 4% (presumably because it wrote the first 4% before the first flashing-status got emitted) and might end at e.g. 98% before the validation process then starts at e.g. 7%. It might be nice if explicit "0% Flashing", "100% Flashing", "0% Validating" and "100% Validating" statuses were written to the DevTools console?

lurch, 2018-02-28:
Another issue that could be merged into #1100 ?

closed 2018-10-15 by jviotti