balena-io/etcher #1693: permission error

csa2k3, 2017-08-14:
- **Etcher version:** 1.1.2
- **Operating system and architecture:** windows 7 64bit
- **Image flashed:**2017-06-21-octopi-jessie-lite-0.14.0
- **Do you see any meaningful error information in the DevTools?**nope

``

lurch, 2017-08-14:
Can you please describe your problem in more detail? i.e. the exact steps you performed, when exactly does the error appear, etc. A screenshot of the error would be useful too.

csa2k3, 2017-08-14:
step1: download it
step2: open it /install it
step3: open it /get the error

lurch, 2017-08-14:
Thanks @csa2k3 that screenshot is helpful.

@jviotti Looks like this is coming from the Windows portion of drivelist https://github.com/resin-io-modules/drivelist/search?utf8=%E2%9C%93&q=permission+error&type=code

csa2k3, 2017-08-14:
how do i fix this ?

github-user-1, 2017-08-20:
I seem to have the same issue - nothing special - just Windows 10, just downloaded and installed Etcher and tried to flash the SD card with Debian Lite from a zip-file. I even tried with running Etcher as Administrator, but that didn't help either unfortunately.

[image]

[image]

jviotti, 2017-08-21:
Can you run the `Etcher.exe` executable from `cmd.exe` and paste the debugging information that should be printed in the console?

github-user-1, 2017-08-22:
Just to let you know - I tried a lot (e.g. including diskpart run as admin to format the card but that all didn't help) so I finally decided to just get me a brand-new SD card. The old one I disposed already and the new one was working fine. Maybe there was just something wrong physicallly with the old one (?).

lurch, 2017-08-22:
Thanks for letting us know @github-user-1 , and glad to hear things are working fine now :+1: 
If you'd still like to try a couple of things on the old card, you could try the steps [here](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives) or you could try using the official SDFormatter tool. But yeah, there's always a possibility that it's physically damaged.

csa2k3, 2017-08-22:
I think ill just go back to win32 at lest it works

lurch, 2017-09-01:
@csa2k3 Please feel free to try helping out on https://github.com/resin-io-modules/drivelist/issues/208 (where there's more detail about this problem)

cron410, 2017-10-03:
If I run ethcher from a Windows 10 x64 admin CMD prompt, i do not get the error and also cannot select a drive to write to.

lurch, 2017-10-12:
@cron410 Have you tried enabling Etcher's "Unsafe mode"? Sometimes removable drives incorrectly get identified as System drives.

sebetovsky, 2017-11-13:
I used Etcher 1.1.2 on Windows 10 x64 and I had the same problem. I realised that it is because of my drive mapping. I used
`subst H: C:\Users\user`

So I solved this by removing this mapping:
`subst H: \D`

Attention after this I had to run the Etcher with Administrator privileges, because automatic privileges aquiration did not work properly.

lurch, 2017-11-14:
@sebetovsky The issue with `subst` drives is being tracked in https://github.com/resin-io-modules/drivelist/issues/158

sebetovsky, 2017-11-14:
@lurch Thanks, but my problem was the error in title and I solved it using deletion of `subst`. Now my situation is solved and I wrote it here only to help others who will have the same problem.

lurch, 2018-02-28:
Could any of the people who previously experienced this problem please re-test with Etcher 1.3.1 ? We've made some significant changes to the way `drivelist` works on Windows.

(Or maybe this should just be closed directly @jhermsmeier ?)

jhermsmeier, 2018-02-28:
Yeah, I'll close this for now; please comment & reopen should this come up again with Etcher v1.3.1+

closed 2018-02-28 by jhermsmeier

csa2k3, 2018-02-28:
Well I fixed my issue all I had to do is uninstall it and re install it
under administrator mode and mine was fix

lfg6000, 2018-06-02:
I had this issue when trying to flash raspbian to a 32gb sandisk sd card on Win10
  You're not authorized to perform this operation
  Please ensure you have necessary permissions for this task

Fixed by temporarily turning off Win 10 Controlled File Access
- open windows defender
- open virus & threat protection
- open ransomware protection
- turn off controlled file access
- Etcher was able to flash the raspbian image to the sd card
- turn controlled file access back on

pandabeard, 2018-12-15:
Hi, 

if somebody is troubleshoot anyway, like me. 
i stopped processes from Acronis and could figure out that, these processes caused access denied problems. This works for me 👍 

Regards Sven

Desoky01, 2019-02-22:
> I had this issue when trying to flash raspbian to a 32gb sandisk sd card on Win10
> You're not authorized to perform this operation
> Please ensure you have necessary permissions for this task
>
> Fixed by [...]

Thank you, you saved my life.

lurch, 2019-02-22:
See also #1960