balena-io/etcher #1697: Etcher-cli crashes on Mac OS right after selecting the SD card reader, but Etcher GUI works fine on the same machine

atn832, 2017-08-17:
- **Etcher version: Etcher-cli-1.1.2-darwin-x64 **
- **Operating system and architecture: Mac OS Sierra 10.12.6 **
- **Image flashed: iot_rpi.img **
- **Do you see any meaningful error information in the DevTools?**

sudo /opt/etcher-cli/etcher /snapshot/etcher/dist/Etcher-cli-1.1.2-darwin-x64-app/lib/cli/etcher.js iot_rpi3.img
? Select drive /dev/disk3 (8.0 GB) - SD Card Reader
? This will erase the selected drive. Are you sure? Yes
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: UNEXPECTED-15
    at readFromSnapshot (pkg/prelude/bootstrap.js:493:16)
    at Object.fs.read (pkg/prelude/bootstrap.js:510:5)
    at ReadStream._read (fs.js:1998:6)
    at ReadStream.Readable.read (_stream_readable.js:348:10)
    at fs.js:1963:10
    at zalgoSafe (pkg/prelude/bootstrap.js:207:10)
    at pkg/prelude/bootstrap.js:424:9
    at FSReqWrap.oncomplete (fs.js:123:15)

Note: Flashing works fine with the Etcher GUI.

atn832, 2017-08-17:
Btw, I just tried to clone the repo, npm install, then run `sudo node etcher.js iot_rpi3.img`, and it worked! Not sure why the public build at https://etcher.io/cli/ failed.

jviotti, 2017-08-21:
Hi @atn832 ,

Thanks for reporting! The issue seems to come from `pkg`, the tool we use to package the CLI as a standalone executable. Would you mind re-posting this issue report at https://github.com/zeit/pkg/issues/new, and ping me so I can stay in the loop?

atn832, 2017-08-23:
Sure @jviotti , I've just filed a ticket there: https://github.com/zeit/pkg/issues/205.

atn832, 2017-08-30:
I get the same error on Windows with https://github.com/resin-io/etcher/releases/download/v1.1.2/Etcher-cli-1.1.2-win32-x64.zip:

```
etcher.exe C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\lib\cli\etcher.js iot_rpi3.img
? Select drive \\.\PHYSICALDRIVE2 (8.0 GB) - Generic- USB3.0 CRW   -SD USB Device
? This will erase the selected drive. Are you sure? Yes
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: UNEXPECTED-15
    at readFromSnapshot (pkg/prelude/bootstrap.js:524:16)
    at Object.fs.read (pkg/prelude/bootstrap.js:541:5)
    at ReadStream._read (fs.js:1998:6)
    at ReadStream.Readable.read (_stream_readable.js:348:10)
    at fs.js:1963:10
    at zalgoSafe (pkg/prelude/bootstrap.js:239:10)
    at pkg/prelude/bootstrap.js:455:9
    at FSReqWrap.oncomplete (fs.js:123:15)
```

jhermsmeier, 2018-03-21:
@atn832 closing this, as the relevant PRs have been merged – next release should be working again :)

closed 2018-03-21 by jhermsmeier

cassianomon, 2018-07-16:
Still happening on v1.4.4 on MacOS