balena-io/etcher #1699: "No removable drive detected." when /tmp is mounted as noexec

szotsaki, 2017-08-19:
- **Etcher version:** 1.1.2
- **Operating system and architecture:** openSUSE Leap 42.3, x64

Mount /tmp as `noexec` (many people and some distros do that by default) and click on "Select drive" after you chose an image. It will show "No removable drive detected".

Console log:
```
Uncaught Error: spawn EACCES
    at exports._errnoException (util.js:1022:11)
    at ChildProcess.spawn (internal/child_process.js:313:11)
    at exports.spawn (child_process.js:399:9)
    at Object.exports.execFile (child_process.js:160:15)
    at Object.module.(anonymous function) [as execFile] (ELECTRON_ASAR.js:200:20)
    at executeScript (/tmp/.mount_npruyq/usr/bin/resources/app.asar/node_modules/drivelist/lib/execute.js:87:16)
    at createTemporaryScriptFile (/tmp/.mount_npruyq/usr/bin/resources/app.asar/node_modules/drivelist/lib/execute.js:153:5)
    at fs.writeFile (/tmp/.mount_npruyq/usr/bin/resources/app.asar/node_modules/drivelist/lib/execute.js:67:5)
    at FSReqWrap.oncomplete (fs.js:112:15)
```

lurch, 2017-08-19:
Yeah, that'll be because we write the `drivelist` script to `/tmp` before executing it. See https://github.com/resin-io-modules/drivelist/blob/master/scripts/linux.sh

We've already moved the drivelist functionality into native code for Windows (because it solved a lot of spawning problems) but we've not yet done that for Linux or macOS.

jviotti, 2017-08-21:
@lurch What do you think about spawning the drives somewhere else in the mean-time, like `$HOME/.etcher/tmp/` or something like that?

lurch, 2017-08-22:
I dunno, that might work. What does @szotsaki think?

I did a bit of googling, and found a bunch of pages telling you how to enable a noexec /tmp directory, and also a bunch of pages telling you how to temporarily disable the noexec option ;-) https://debian-administration.org/article/57/Making_/tmp_non-executable

I did a little bit of digging, and `drivelist` uses https://nodejs.org/api/os.html#os_os_tmpdir to determine where to write its files, and it seems that function respects the TMPDIR variable:
```
$ node
> os.tmpdir()
'/tmp'
$ TMPDIR=/foo node
> os.tmpdir()
'/foo'
$ node
> os.tmpdir()
'/tmp'
```

So @szotsaki could you try running `TMPDIR=some/executable/path etcher-electron` (or maybe `TMPDIR=some/executable/path etcher-1.1.2-linux-x86_64.AppImage`) and let us know if that fixes your problem?

szotsaki, 2017-08-22:
Yes, I used the `remount` option to make /tmp executable but only `root` can do that, so I think it's not a good solution. Modifying it in a program and knowing that that program can accidentally crash leaving /tmp in `exec` state may have awful consequences.

Directory selection idea: I recommend to adhere to [Freedesktop specification](https://specifications.freedesktop.org/basedir-spec/latest/ar01s03.html), specifically `$XDG_RUNTIME_DIR` or maybe `$XDG_CACHE_HOME`.

Running ethcher with `TMPDIR=$XDG_RUNTIME_DIR ./etcher-1.1.2-x86_64.AppImage` made it work with `noexec`, thank you for the tip.

lurch, 2017-08-22:
> Modifying it in a program

Sorry, that's not what I was suggesting. I was just illustrating that other programs can have problems with a non-executable `/tmp` directory too ;-)

Thanks for the info about `$XDG_RUNTIME_DIR` !

jviotti, 2017-08-22:
@szotsaki Thanks a lot for the XDG suggestions. I'll be sending a PR very soon :)

jviotti, 2017-08-22:
See https://github.com/resin-io-modules/drivelist/pull/209

lurch, 2017-10-05:
Re-opening this issue, as the proposed fix actually _breaks_ things for other Linux distros, see https://github.com/resin-io-modules/drivelist/pull/209#issuecomment-334443349

(BTW it seems `$XDG_CACHE_HOME` isn't set on Ubuntu 14.04)

lurch, 2017-10-10:
Closing, as drivelist in Etcher has been upgraded.

closed 2017-10-11 by lurch