balena-io/etcher #1700: SSL Issue when booting up

eveaide, 2017-08-19:
- **Etcher version:1.1.2**
- **Operating system and architecture:Win10 Insider Preview Version 1703 (Build 16257.1) x64**
- **Image flashed:N/A**
- **Do you see any meaningful error information in the DevTools**
[image]

I can gather it's some form of SSL issue, same happens on Portable and Installed. When booting, these error messages come up;
[image]
[image]

lurch, 2017-08-20:
It looks like you're actually running into two separate problems here - the first is that Etcher couldn't scan your drives, and then when Etcher tried to report that error back to our reporting service, you then ran into the second SSL-related error.

Are you using BitLocker? Maybe the first problem is a duplicate of #1687 ?

eveaide, 2017-08-20:
No Bitlocker, but I may have resolved this anyway. After shutting down "360 Total Security", it runs fine now. They must of updated some blacklist rules on their end, as it used to work fine before with both Etcher and 360TS.

jviotti, 2017-08-21:
@eveaide Are both issues solved when you shut down 360 Total Security? Can you actively reproduce by starting the software again?

lurch, 2017-08-21:
@jviotti I suspect that whatever @eveaide did to fix the "Couldn't scan drives" error (perhaps that was indeed turning off "360 Total Security") also meant that there was then no error message to report to our reporting service, and thus we have no way of knowing if that also fixes the SSL error or not ;-)

lurch, 2018-02-28:
@eveaide Are you still experiencing these problems with Etcher 1.3.1 ?

closed 2018-10-15 by jviotti