balena-io/etcher #1701: Download failure on Windows

gopal-amlekar, 2017-08-21:
Hi,
I am trying to download Etcher tool on Windows 10 64 bit OS. The problem is the download fails after a while and reason shown by google chrome is "Forbidden". I then tried to download it via a download package manager. It asks for authentication when the download reaches around 40% or so. 
I have tried to download both the portable and installer files. The paths are:
https://github.com/resin-io/etcher/releases/download/v1.1.2/Etcher-Portable-1.1.2-x64.exe
https://github.com/resin-io/etcher/releases/download/v1.1.2/Etcher-Setup-1.1.2-x64.exe

These paths are taken from the website https://etcher.io. They are also same as those appearing on this page - https://github.com/resin-io/etcher/releases. 
Strangely, I couldn't find any 'releases' folder on github or probably they are private hidden paths?

I then tried to use my github username/password when asked for authentication. It did then complete the download but when I run the setup file now, it shows an error: Installer integrity check has failed.  Common causes include incomplete download or damaged media. 

More details:

- **Etcher version:** - V1.1.2 (based on the filename)
- **Operating system and architecture:** - Windows 10 64 bit
- **Image flashed:** - None
- **Do you see any meaningful error information in the DevTools?** - No since the tool is not yet installed.

Am I missing something?

Thanks a lot,
Gopal

lurch, 2017-08-21:
That's really weird - both those links download fine for me without any errors or authorisation requests (I tested both in an incognito Chrome window and with `wget`).
Are you behind a corporate proxy or something, which might be interfering with your downloads? Or perhaps you have a virus checker or something which is mistakenly blocking them?

gopal-amlekar, 2017-08-21:
Well yes weird so I couldn't think of any clue. No I am not behind any proxy or firewall. It is my personal laptop. But yes an antivirus is there. Don't want to disable antivirus even for a minute really :)
I will try a few more alternatives like Firefox or another machine running Ubuntu etc.

lurch, 2017-08-21:
Thanks, please let us know what you discover :)

gopal-amlekar, 2017-08-21:
Update:
I could finally download the setup file and could install the tool on Windows. Didn't do anything different. Just retried few more times via the download manager used earlier.
 
The portable one still asks for authentication and the name of site shown is: github-production-release-asset-2e65b.s3.amazonaws.com

lurch, 2017-08-21:
Closing this as an intermittent issue with @gopal-amlekar 's ISP / internet-connection.

(although feel free to add further comments if you discover anything else)

lurch, 2017-08-21:
Pinging @craig-mulligan in case he's ever seen anything like this before?

closed 2017-08-21 by lurch

jhermsmeier, 2017-08-29:
> the name of site shown is: github-production-release-asset-2e65b.s3.amazonaws.com. On providing github authentication, it completes the download

This sounds *really weird* – why would anything ask you for your GitHub credentials for a download that's public?

> fails to start with some error message: The application failed to start because its side by side configuration is incorrect.

This appears to hint at corrupted binaries; meaning that either the download got currupted due to network issues, or Chrome itself (or another part of the system) might have a part of it corrupted.

You can run `sfc /scannow` (System File Check), and / or `dism /online /cleanup-image /scanhealth` to check if Windows can find anything. In case it does, `dism /online /cleanup-image /restorehealth` should attempt to repair anything it can within the system – other issues would prob. have to be resolved manually (i.e. by re-installing Chrome).

gopal-amlekar, 2017-08-30:
I couldn't think of any reason or logic behind asking the github authentication.. So that still is a mystery for me.

But the other issues probably my laptop had gone crazy. Didn't give any meaningful result with the dism or sfc tools. For various other reasons, I had to completely format it and reinstall Windows on it. 

So I can only blame my laptop / OS / Chrome for the corrupt binaries.

jhermsmeier, 2017-08-30:
Oh, sorry to hear that, but thanks for the update!