balena-io/etcher #1706: "Could not locate the bindings" on Windows 10

atn832, 2017-08-23:
- **Etcher version: \Etcher-cli-1.1.2-win32-x64**
- **Operating system and architecture: Windows 10**
- **Image flashed: iot_rpi3.img **

I downloaded the build for Windows x64 from the etcher/cli website (https://etcher.io/cli/, https://github.com/resin-io/etcher/releases/download/v1.1.2/Etcher-cli-1.1.2-win32-x64.zip). I extracted the zip to C:\etcher-cli like in the instructions, and also added that path to my PATH environment variable. Etcher-cli crashes before even listing the available drives to flash.

```
C:\etcher-cli>etcher
Usage: etcher.exe C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\lib\cli\etcher.js [options] <image>

Options:
  --help, -h     show help  [boolean]
  --version, -v  show version number  [boolean]
  --drive, -d    drive  [string]
  --check, -c    validate write  [boolean] [default: true]
  --yes, -y      confirm non-interactively  [boolean]
  --unmount, -u  unmount on success  [boolean] [default: true]

Examples:
  etcher.exe C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\lib\cli\etcher.js raspberry-pi.img
  etcher.exe C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\lib\cli\etcher.js --no-check raspberry-pi.img
  etcher.exe C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\lib\cli\etcher.js -d /dev/disk2 ubuntu.iso
  etcher.exe C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\lib\cli\etcher.js -d /dev/disk2 -y rpi.img

Exit codes:
  0 - Success
  1 - General Error
  2 - Validation Error
  3 - Cancelled

If you need help, don't hesitate in contacting us at:

  GitHub: https://github.com/resin-io/etcher/issues/new
  Gitter: https://gitter.im/resin-io/etcher

Missing image

C:\etcher-cli>etcher C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\lib\cli\etcher.js iot_rpi3.img
Could not locate the bindings file. Tried:
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\build\drivelist.node
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\build\Debug\drivelist.node
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\build\Release\drivelist.node
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\out\Debug\drivelist.node
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\Debug\drivelist.node
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\out\Release\drivelist.node
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\Release\drivelist.node
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\build\default\drivelist.node
 → C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\compiled\6.10.3\win32\x64\drivelist.node

C:\etcher-cli>
```

atn832, 2017-08-23:
Note: I get a similar error with the x86 build.

jviotti, 2017-08-28:
Hi @atn832 ,

Thanks for the heads up. Turns out that `pkg` creates an application snapshot based on the current working directory, which is messing up the paths to the add-ons.

I'm fixing it as I write. For the time being, you can easily run the CLI from source, which should work fine:

```
git clone https://github.com/resin-io/etcher
cd etcher
npm install --build-from-source
node ./bin/etcher
```

jviotti, 2017-08-28:
@atn832, I re-published the Windows CLI packages. Can you try https://github.com/resin-io/etcher/releases/download/v1.1.2/Etcher-cli-1.1.2-win32-x64.zip again?

atn832, 2017-08-30:
Hi @jviotti , now it runs and lets me select the drive to flash. However when it tries to write on the drive, it crashes with the same error as on Mac OS (https://github.com/resin-io/etcher/issues/1697, https://github.com/zeit/pkg/issues/205):

```
etcher.exe C:\snapshot\etcher\dist\Etcher-cli-1.1.2-win32-x64-app\lib\cli\etcher.js iot_rpi3.img
? Select drive \\.\PHYSICALDRIVE2 (8.0 GB) - Generic- USB3.0 CRW   -SD USB Device
? This will erase the selected drive. Are you sure? Yes
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: UNEXPECTED-15
    at readFromSnapshot (pkg/prelude/bootstrap.js:524:16)
    at Object.fs.read (pkg/prelude/bootstrap.js:541:5)
    at ReadStream._read (fs.js:1998:6)
    at ReadStream.Readable.read (_stream_readable.js:348:10)
    at fs.js:1963:10
    at zalgoSafe (pkg/prelude/bootstrap.js:239:10)
    at pkg/prelude/bootstrap.js:455:9
    at FSReqWrap.oncomplete (fs.js:123:15)
```

atn832, 2017-08-30:
Closing this as the bindings issue has been fixed.

closed 2017-08-30 by atn832