balena-io/etcher #1711: Scanning for drives fails completely due to unlocked Bitlock-To-Go driver

soletan, 2017-08-25:
- **Etcher version:** 1.1.2
- **Operating system and architecture:** win32 x64
- **Image flashed:** none
- **Do you see any meaningful error information in the DevTools?**

{"stack":"Error: Couldn't scan the drives: An unknown error occurred","message":"Couldn't scan the drives: An unknown error occurred"}

console.(anonymous function) | @ | C:\Users\cepharum\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven…:51

On detaching locked drive scanning drives succeeds without error.

jviotti, 2017-08-28:
Hi @soletan,

Check the following ticket: https://github.com/resin-io/etcher/issues/1687. Can you try the development build published there, and confirm it works for you?

closed 2017-08-28 by jviotti

soletan, 2017-08-28:
Tried the version attached to that ticket (haven't seen it before, sorry) and it works in my case as well.