balena-io/etcher #1716: Can Etcher be used to customize the base Raspbian image?

zharvey-welltok, 2017-08-29:
I was wondering if anyone in this community has used Etcher to burn OS images to a Raspberry Pi. I have a Rapberry Pi 1 Model A and would like to use Etcher to burn [Raspbian Stretch Lite](https://www.raspberrypi.org/downloads/raspbian/) to it with some custom modifications:

* I want to start with Raspbian Stretch Lite as the base image
* Then I have several *nix packages that I'd like to add to that image using something like `apt-get`
* Then I want to install my own app/service at a particular location on the file system
* Finally I want the whole OS configuration/image to be burned onto the pi so that when I power it on, Raspbian starts running and its just like I had manually provisioned the server myself

Can Etcher accomplish all this for me? Or do I need to load the Raspbian image into a different tool and customize it (run `apt-get`, manually install my app, etc.) from there? What would that tool pipeline look like? Thanks in advance!

jviotti, 2017-08-29:
We are planning some basic configurations features (https://github.com/resin-io/etcher/issues/718), but for the time being, I guess the best approach would be to mount the ISO on a GNU/Linux system, enter the OS context using something like `chroot`, make your modifications, and then flash the ISO using Etcher as many times as you like :)

jhermsmeier, 2017-08-29:
Short answer: Nope, not at this time.

Etcher flashes images byte-by-byte at the moment, so any pre-configuration / modifications to the image itself have to be done to the image prior to flashing.

> Then I have several *nix packages that I'd like to add to that image using something like apt-get

I'm not sure what one would use to do that, but there surely is something out there.
You might want to have a look at https://github.com/resin-io/reconfix (which is still WIP though).

> Then I want to install my own app/service at a particular location on the file system

If you want to run resinOS on that Pi, we have https://github.com/resin-io/resin-preload, and the resin-cli which can preload apps into the OS. If you're not planning to do that, it should be as easy as mounting the respective file system from the image and copying the files over.

lurch, 2017-09-04:
> I was wondering if anyone in this community has used Etcher to burn OS images to a Raspberry Pi.

Yes, many **many** times ;-)  We're even included in the Raspberry Pi Foundation's own [instructions](https://www.raspberrypi.org/documentation/installation/installing-images/README.md).

I think what you're hoping to achieve falls much more under the remit of http://pibakery.org/ than it does Etcher. PiBakery is totally Pi & Raspbian specific, whereas Etcher is a more generic tool and so works with other distros / boards / OSes too.

I'm not sure what you mean by "I want the whole OS configuration/image to be burned onto the pi" - you don't burn images to the Pi itself (the Pi has no internal storage) - you flash the image onto an SD card (using your PC), and then put the SD card into the Pi before then connecting power to the Pi.

If PiBakery doesn't quite do what you want, the _easiest_ way to do what you appear to be attempting to do (which AFAIK is what most people do) would be to just burn the Raspbian image as normal to an SD card, boot it up in the Pi, make your desired modifications, power down the Pi, and make a backup of the SD card on your PC with e.g. Win32DiskImager or dd (this is a feature not yet supported by Etcher #266 ).
If you really want to "get deep" you could look at customising https://github.com/RPi-Distro/pi-gen which is the tool used to produce the Raspbian images in the first place.

I'm going to close this issue, as I think installing custom packages is well beyond the scope of Etcher.

closed 2017-09-04 by lurch