balena-io/etcher #1718: etcher spies on the user without consent, and transmits the spying results unencrypted

sneak, 2017-08-30:
- **Etcher version:**

1.1.2

- **Operating system and architecture:**

osx 10.12.6 amd64

[image]

I did not consent or opt in to such tracking.  To add insult to injury, you are transmitting the results of such unethical tracking unencrypted, violating every best practice about the safe handling of user data there is.

jviotti, 2017-08-30:
Hey there @sneak ,

These logged events are for error tracking purposes. They are anonymous and you can disable them through the settings page.

Regarding HTTP, I was not aware that the Mixpanel client was doing so. Do you have any clues @Shou ?

jhermsmeier, 2017-08-30:
I was just looking into it, and the Mixpanel client uses HTTP by default, which we don't override in corvus. Mending that at the moment.

closed 2017-09-08 by jhermsmeier