balena-io/etcher #1719: Etcher doesn't detect named USBs

ajmeese7, 2017-09-04:
- **Etcher version: 1.1.2**
- **Operating system and architecture: Windows 10 64-bit**

I noticed Etcher won't detect any named USB. By named I mean the user went into the USB properties and changed the name. I have tried with multiple USBs and it doesn't detect any of them.

lurch, 2017-09-04:
I'm not sure what you mean. Could you add some screenshots to show examples?
Does Etcher detect "the USB" (whatever you mean by that) before you rename it?

ajmeese7, 2017-09-04:
I'm not sure how the term USB is ambiguous but here is a screenshot of Etcher only detecting the unnamed flash drive.

[image]

lurch, 2017-09-04:
Thanks, that screenshot clarifies that you're actually talking about USB flash drives. (I wasn't sure if you might have been talking about USB ports, USB interface cards, USB drivers, etc.)
I'll try to reproduce and investigate tomorrow.

lurch, 2018-02-28:
@ajmeese7 Do things now work properly in Etcher 1.3.1 ?

jhermsmeier, 2018-02-28:
These names you're referring to are volume labels, and are part of the file system(s) stored in the partitions on the device, or in the partition table (for GUID Partition Tables and Apple Partition Maps), not a name of the device itself – we do list devices with their volume mountpoints (i.e. F: on Windows), but don't read the partition or volume labels at the moment.

closed 2018-02-28 by ajmeese7

ajmeese7, 2018-06-01:
I just checked in 1.4.4 and it properly reads the USB drives now