balena-io/etcher #1721: Add high contrast mode or tweak text contrast

jhermsmeier, 2017-09-06:
The current text contrast is pretty hard to read for people with impaired vision (and / or lower contrast screens, or unforgiving lighting conditions) – we should think about either adding a high contrast mode to be enabled through the settings, or bumping the overall text contrast.

jhermsmeier, 2017-09-07:
See also: https://github.com/electron/electron/pull/7518

lurch, 2017-09-12:
Probably something for @konmouz to look at?

konmouz, 2017-09-12:
@jhermsmeier, can you post an example screen shot? The overall contrast ratio should be pretty good at the moment. Is it about the light grey text on the dark grey background?

jhermsmeier, 2017-09-12:
[image]

Simulated by upping the contrast here, half the UI becomes nearly invisible. I'm not sure if there are any good vision impairment simulations we could apply to get a better picture of things that need attention? There's also a description from a user in the linked Front convo, detailing their difficulties.

lurch, 2018-02-28:
Related to resin-io/hq#1024

closed 2018-10-15 by jviotti