balena-io/etcher #1722: Invalid partition table - recursive partition on /dev/sdb

JustAnotherJack, 2017-09-07:
- **Etcher version: 1.12.
- : Ubuntu 16.04.1 LTS.  Mate.  
-  Linux Mint 18.2. 
8Gb Usb.  Operational.. Previously had another Distro. Used Gparted and Formatted to Fat 32
Etcher says  indicates sucessful Install.
USB does not show up in file manager.  Gparted results in: WARNING:   Invalid partition table - recursive partition on /dev/sdb.
Thank you in advance.

jhermsmeier, 2017-09-07:
Hey @JustAnotherJack – which image exactly were you flashing, and which format is it in (img, iso)?

JustAnotherJack, 2017-09-07:
Hi,  and thank you.

linuxmint-18.2-cinnamon-32bit  iso.

lurch, 2017-09-12:
I haven't checked yet, but it wouldn't surprise me if this is just another instance of https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

@JustAnotherJack Ignoring the errors from the file manager and GParted, does your USB drive actually boot?

JustAnotherJack, 2017-09-12:
> DOCUMENTATION.md#flashing-ubuntu-isos

No. Does not.
I made one using the command line, and it boots fine.
Thanks

jhermsmeier, 2017-09-13:
@JustAnotherJack what did you run on the command line?

JustAnotherJack, 2017-09-14:
I don't remember the exact string, but it was the "dd" command that worked.

lurch, 2017-09-15:
Both `dd` and `Etcher` do exactly the same thing - copy the disk image exactly byte-for-byte onto your destination drive. So I've got absolutely no idea why you'd be seeing different behaviour from the two tools :-/

JustAnotherJack, 2017-09-15:
I'll try again with the same distro (Manjaro), and another.
Thanks

lurch, 2017-09-18:
Hi @JustAnotherJack , did you manage to make any progress?

JustAnotherJack, 2017-09-18:
Yes. As I said, when I used the command line it worked.  I can do other
distros using etcher, but not Manjaro.

lurch, 2017-09-19:
@JustAnotherJack Can you please confirm which distros (and which versions of those distros) are and aren't working? Earlier on you were talking about Linux Mint, and now you're talking about Manjaro...?

JustAnotherJack, 2017-10-02:
My apologies.  The Manjaro comment was meant for another forum.

closed 2017-12-04 by jviotti