balena-io/etcher #1723: Guix SD .xz file not detected

kwill, 2017-09-07:
- **Etcher version:** 1.1.2
- **Operating system and architecture:** Ubuntu 17.04 x64
- **Image flashed:** guixsd-usb-install-0.13.0.x86_64-linux.xz
- **Do you see any meaningful error information in the DevTools?** Yes, see below.

**Steps to reproduce:**

* Download Guix SD image: https://www.gnu.org/software/guix/
* Open Etcher
* Click "Select image"
* Select Guix SD file

**What should happen?**

Image is loaded. Flash process continues.

**What happens instead?**

After a significant pause the following error message appears:

Error opening image
Something went wrong while opening guixsd-usb-install-0.13.0.x86_64-linux.xz
Error: File format not recognized

And the following error is reported in devtools:

<code>
{"name":"LZMA_FORMAT_ERROR","message":"File format not recognized","cause":{"code":7,"name":"LZMA_FORMAT_ERROR","desc":"File format not recognized"},"isOperational":true,"stack":"LZMA_FORMAT_ERROR: File format not recognized\n    at /opt/Etcher/resources/app.asar/node_modules/lzma-native/index.js:521:18\n    at tryCatcher (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise.successAdapter [as _fulfillmentHandler0] (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/nodeify.js:23:30)\n    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)\n    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._resolveCallback (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)\n    at Promise._settlePromiseFromHandler (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)\n    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._resolveCallback (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)\n    at Promise._settlePromiseFromHandler (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)\n    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at /opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/nodeback.js:42:21\n    at FSReqWrap.wrapper [as oncomplete] (fs.js:603:17)","code":7,"desc":"File format not recognized"}
</code>

jhermsmeier, 2017-09-07:
There's definitely something fishy going on there; I do see different devtools output though – @kwill is your devtools output really from Etcher v1.1.2?
```
Invalid image ({
  "path": "/Users/Jonas/Downloads/guixsd-usb-install-0.13.0.x86_64-linux.xz",
  "archiveExtension": "xz",
  "extension": "x86_64-linux",
  "size": {
    "original": 198805472,
    "final": {
      "estimation": false,
      "value": 1288490496
    }
  },
  "hasMBR": true,
  "hasGPT": false,
  "partitions": [{
    "type": 131,
    "id": null,
    "name": null,
    "firstLBA": 2048,
    "lastLBA": 2416231,
    "extended": false
  }, {
    "type": 239,
    "id": null,
    "name": null,
    "firstLBA": 2416231,
    "lastLBA": 2498152,
    "extended": false
  }, {
    "type": 0,
    "id": null,
    "name": null,
    "firstLBA": 0,
    "lastLBA": 0,
    "extended": false
  }, {
    "type": 0,
    "id": null,
    "name": null,
    "firstLBA": 0,
    "lastLBA": 0,
    "extended": false
  }]
})
```

jhermsmeier, 2017-09-08:
The bug here is that Etcher expects two file extensions (i.e. `.img.xz`) for flat compressed files (i.e. xz, bz2, gz, etc) – and fails to handle it correctly (assume a flat blob) if it's missing.

For now, to work around this, you should be able to rename your image from
```
guixsd-usb-install-0.13.0.x86_64-linux.xz -> guixsd-usb-install-0.13.0.x86_64-linux.img.xz
```
and then that should work as expected.

kwill, 2017-09-08:
Definitely 1.1.2, but I was using the decompressed file :P 

Adding the second suffix solved the real problem, thanks!

closed 2017-09-08 by kwill