balena-io/etcher #1725: Etcher Flahsing Validation Failed

Alpha-Phoenix, 2017-09-09:
- **Etcher version: 1.1.2**
- **Operating system and architecture: Windos 10 x64 Build 10.0.15063**
- **Image flashed: [RASPBIAN STRETCH WITH DESKTOP](https://www.raspberrypi.org/downloads/raspbian/)**
- **Do you see any meaningful error information in the DevTools? It can not open**

 I'm having problems when flashing raspbian stretch image into my **Raspberry pi zero w**. I tried to flash on linux and windows with various sd cards (8 GB | 16 GB) but the problem persist. This is the error message shown by etcher.
[image]

How can I fix this?

lurch, 2017-09-12:
Have you got a different SD card-writer you can test with?

NateZimmer, 2017-09-13:
Same issue w/ a brand new SD card & brand new reader.

jhermsmeier, 2017-09-13:
Might be with the zip handler (or the zip file) – will give this a run and check.

Abrastreve, 2017-09-13:
Same problem here. On Windows 10 I tried several times to flash Jessie light onto two different SD cards (one of which brand new) with Etcher-Portable-1.1.2-x64. Each time I get the above potential corruption message.

jhermsmeier, 2017-09-13:
Are you all on Windows 10 Insider Preview builds?

Abrastreve, 2017-09-13:
No, I do not use an Insider Preview version but a regular customer release of W10

Alpha-Phoenix, 2017-09-13:
> Might be with the zip handler (or the zip file) – will give this a run and check.

@jhermsmeier I think this is not the problem. I did checksums (SHA256) and the generated code is the same shown on the raspbian download webpage, and I'm using the latest version of 7-Zip. I tried to flash with Win32DiskManager. The flashing proccess looks well done but my raspberry pi zero w dont boot.

jhermsmeier, 2017-09-13:
@Alpha-Phoenix wait, you flashed the same image with Win32DiskManager, and it still didn't boot?

jhermsmeier, 2017-09-13:
So, I just tested this on Windows 10 build 14393 and the current fast-ring Win 10 Insider Preview build, and the RPi Zero W I have here boots – can't reproduce this so far.

@NateZimmer @Alpha-Phoenix what kind of SD-card reader are you using (make & model, or is it built-in)?

NateZimmer, 2017-09-14:
I am using a 'Digipower 42-in-1 Card Reader' USB to SD card(walmarts best).

I just tried it again with a different brand new SD card on my HP Printer(which has a built in USB->SD) and it **worked**. Either the prior card or the prior usb->sd reader is suspect. Will update this once I figure out which.

Alpha-Phoenix, 2017-09-14:
I can't find my SD card reader (model or maker), but I've tried to flash using an usb adapter and the problem still happening.

I'm using a [ATIV Book 2.4 270E4E-KD5 14""](http://www.samsung.com/br/support/model/NP270E4E-KD5BR)

Abrastreve, 2017-09-17:
My trials were done on the internal cardreader/writer of a Toshiba SatelliteC50 laptop using Windows10 build 15063.

AllistonJohn, 2017-09-24:
I too am getting this message. Windows 10. Tried 3 different new SD cards.

NateZimmer, 2017-09-24:
@AllistonJohn for me it was a bad SD card reader. I'd recommend trying a different SD card reader.

AllistonJohn, 2017-09-24:
Ok I will try that! Thank you.

John

maroquio, 2017-09-27:
Same problem here, with a internal SD reader of a 2016 Dell notebook and with a SD reader in my desktop. If I made an image of an old card with Emlid Raspbian and flash it, no error occurs. Someone know what is happening? I'm using Windows 10 with latest updates.

AllistonJohn, 2017-09-28:
I tried different cards, and a different computer, a different formatter,
and a different flash utility and still got an error. I continued on and
used the card as  is.

jhermsmeier, 2017-09-28:
@AllistonJohn @maroquio which image were you two flashing?

ankitume, 2017-10-14:
I also ran into trouble while writing the Raspbian image to the Kingston 8 GB card. Etcher takes lot of time validating.

fvilers, 2017-10-26:
I also ran into this issue, 4 of 6 SanDisk cards (16gb) where written with a validation failure. The file is 2017-09-07-raspbian-stretch-lite.img. I'm going to boot on them later one, hoping they will be ok. As something mentioned before, flashing with Win32DiskImager-0.9.5 doesn't gave any error (event with MD5 checked). Card read in Lenovo T460s built-in.

fvilers, 2017-10-26:
Guess what? I successfully flashed all my cards with the laptop running powered on the sector. It seems like Windows 10 battery saving can cause issue with the flashing process, maybe a read or write rate not high enough.

maroquio, 2017-10-26:
I burned it successfully in Linux command line and it’s working normally.
Regards,
Ricardo Maroquio

closed 2017-12-04 by jviotti

gocursor, 2018-01-18:
I have the same problem with a brand new SanDisk USB stick. Maybe this is because Windows is creating the "System Volume Information" hidden folder just after write is completed and volume is mounted - and before verification is done...

gorbehnare, 2018-10-27:
I think this is definitely Etcher issue. I use the internal PCIe SD card reader on my HP Elitebook 8460p and using ANY SD card I get the message trying to image 2018-10-09-raspbian-stretch-lite.img to SD. 
Indeed the files in the SD are corrupted as Raspbian fails to works properly and many things fail on boot. 
However using the same MicroSD and card reader I can image th SD using Win32DiskImager v1.0.0 and everything works just fine. 
This is not a matter of verification failing, Etcher is actually writing corrupt data to SD card.

jungerm2, 2019-08-11:
I had this problem too. I've tried flashing 2 brand new SD cards, and one new USB flash drive, al of which failed. I tried toggling all the windows power features on and off as well. After a couple of hours of frustration I decided to try another flashing software, Rufus worked in the first go. Although I really like etcher, I need it to work!

rthomas67, 2019-08-17:
I know this was closed a long time ago, but etcher still seems to have issues with a certain combination of OS and USB-flash-reader/writer hardware.  Just adding more info in case it helps anyone get around this.

Just downloaded etcher 1.5.45 portable (on Windows 10).  It is failing on a DigiPower Multi Card Reader - Model No: DP-MCR4 (reader/writer works fine otherwise) with a Sandisk 32GB Ultra microSDHC UHS-I card.  Not getting the popup message, but there is a subtle, almost unnoticeable, red dot on the final screen that says "1 Failed Device".  When I hover the mouse pointer over that I get an explanation that the "Source and destination checksums do not match"  Same failure occurred for both the zip of 2019-07-10-raspbian-buster-full (~2.4 GB), and the unpacked .img file (~6.3 GB).  The files are stored on an NTFS drive, so file size shouldn't be an issue.  The .img file was unzipped using 7zip so the zip64 format shouldn't have mattered.

One thing that _might_ be interfering with etcher is Windows' disk/drive management.  After the flash/validation process is complete, Windows displays a "Format disk" prompt saying that the flash drive must be formatted before it can be used.  Clicking cancel presumably dismisses that prompt without Windows taking any action.  But, _if_ that could be disrupting the validation/checksum process, maybe there is a way etcher could prevent Windows from attempting to re-mount the drive/filesystem until the validation is complete (if so, someone could open a new github issue for that). 

After etcher flash/validation, and after disconnecting and reconnecting the USB flash reader + card, Windows shows that there is a valid boot partition on the microsd card.  The card still boots up, resizes the filesystem, initiates the Raspbian Buster setup process, etc.  I still have concerns using etcher (with this combination of OS and reader/writer hardware) that there will be something corrupted on the flash drive and I won't find out what it is until much later, so unless I can dig up another reader/writer that works, I'm with @jungerm2 ... can't use Etcher right now.
 
[image]

lurch, 2019-08-18:
@rthomas67 See #2210 ?

doctorxyz, 2019-09-08:
Sorry about resurrecting this old thread, but I agree with @fvilers!
It seems my Windows 10 Notebook only corrupts SDCards only when I run the Balena Etcher flashing/validating processes depending on battery...! The issue does not happens when my Notebook is being powered from power switch.
Wouldn't it be time for calling testers to check this scenario? After receiving enough feedback, the Balena Etcher development team could add an alert / warning message, right before the process begins.

necrose99, 2020-02-28:
2 weeks ago working windows update borked.

dundir, 2020-03-27:
I got this error on W10 1903 as well using the same 42-in-1 reader with an ONN Micro SD card. 

Digging into it a bit deeper this looks like a low level problem with either the reader, writer or both and not etcher. 

I ran some benchmarks on linux with dd instead and what appears to be happening is it will be able to read up to some point and then all the blocks after that throw a buffer_IO error on async page write.

Sometimes it can read/write with no issues (briefly), quite a lot of the time it fails. 

Also, just for giggles I bought another of the same brand (from Walmart) and get the exact same issue when swapping either the memory card or the reader, so it looks like there is a flaw. 

Maybe if I can find some time in the next millenia I'll dig into it with an oscilloscope.

ZurMaD, 2021-05-11:
Same checksum error with different ISOs on different USB ports with differents USB. Definitely it's a software error but don't know what's happening. I tried with an AC power plugged and didn't work, and also without AC Power plugged and 

I'm on Ubuntu 20.04 and in the past I used balena a lot of times for different ISOs without problems. However, now any ISO validation finish in ckecksum error.

Without validation, installing Ubuntu on the SSD gives an error of files corrupted. What's happening?

[image]

lurch, 2021-05-12:
@ZurMaD It might be worth opening a new issue, rather than adding comments to an already-closed issue.

lurch, 2021-05-12:
@dundir It seems those cheap XX-in-1 card readers tend to be generally unreliable, see e.g. https://github.com/raspberrypi/rpi-imager/issues/212

n057r4d4mu5, 2021-12-31:
All Microsoft product is a useless sh..t, it is fact, but this error not _unconditionally_ Windows 10 bug, because i use Debian Linux, but i got same issue.