balena-io/etcher #1727: tried using this on windows to make a usb iso from ubuntu 17 but

foxjazz, 2017-09-15:
results in not reading disk.
latest version of etcher was used running windows 10
it says it's done 

used disk part to fix the usb drives.

jhermsmeier, 2017-09-15:
Do you mean that Windows cannot read the disk? Or that yor computer can't boot from it?

lurch, 2017-09-19:
Unless the USB disk is actually unbootable, I suspect this is just the standard 'problem' of Windows not being able to read Linux-format partitions. Either that, or https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

@jhermsmeier The problem of Windows not displaying Linux-disks has come up a number of times before - maybe worth adding something about this to the documentation?

closed 2017-12-04 by jviotti