balena-io/etcher #1728: Sincerely recommend to add a funtion: Avoid flashing your removable hard disk driver.

kevinkm, 2017-09-15:
I was too stupid to remember I connected my removable hard disk and USB sticker at the same time, therefore, as you may guess, I just pressed "flash" as usual as I do.......wellllllll, now Iam recovering datas from hard disk driver with tremendous self-accusation.

So, Iam thinking when etcher has checked the target driver is bigger than 100GB, it ought to force user to type some words like "I will flash my hard disk".

I have to say this is my first time to make a huge mistake flash a wrong driver, since I believe etcher is smart enough to be trusted.

lurch, 2017-09-19:
We've had suggestions like this before... the problem always comes down to where to draw the "cutoff point" between Hard Drives and SD cards - for example I personally own an 80GB external HDD, and a 200GB micro SD card.
Etcher __does__ clearly show the size of each drive in the drive-selection window, and also in the main window after selecting a drive. I think it's always a very delicate balancing act between making something "safe" and making something "have too many annoying pop-ups which people will just ignore anyway".

alexandrosm, 2017-09-21:
Perhaps a warning for drives over X size (64gb?) Something like -- "This
drive looks quite large. Please make sure you've not accidentally selected
a removable hard drive. [All OK, Continue] [Abort Flashing]"

cc @jviotti @jhemsmeier

jviotti, 2017-09-22:
I think that's a great idea! 👍

—
Juan Cruz Viotti,Software Engineer, Resin.io (https://resin.io)
twt: @resin_io

--- original message ---

—

--- end of original message ---

mverleun, 2017-10-23:
I'm careful when writing images to the SSD card. Especially since I've got a 256Gb Transcend Jetdrive that I use for data storage.
This drive always shows up as a possible target to write images to. 
Is it possible to (manually) create a blacklist of targets that Etcher won't show?

lurch, 2017-10-24:
I hadn't heard of that before, so just googled and found https://www.amazon.co.uk/Transcend-256GB-JetDrive-Storage-Expansion/dp/B01BE0C2PU
Looks like it's basically a half-size SD card? (and if it plugs into the Mac's SD slot, then it's no surprise that Etcher detects it as a removable drive!)
Could you send us your `diskutil` output, so that we can see if there's a way to uniquely identify this particular drive?

mverleun, 2017-10-24:
It's a nice way to add additional storage. It's indeed a half size SD card, directly inserted into the designated slot.
Identifying this drive from diskutils output is not easy as far as I can tell.  The only recognisable attribute is it's name, which is chosen when formatting the device. See disk utils output below:
```
$ diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *480.1 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:                 Apple_APFS Container disk1         479.1 GB   disk0s2

/dev/disk1 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +479.1 GB   disk1
                                 Physical Store disk0s2
   1:                APFS Volume Macintosh HD            442.2 GB   disk1s1
   2:                APFS Volume Preboot                 20.9 MB    disk1s2
   3:                APFS Volume Recovery                519.9 MB   disk1s3
   4:                APFS Volume VM                      2.1 GB     disk1s4

/dev/disk2 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *259.7 GB   disk2
   1:                  Apple_HFS Transcend JetDrive      259.7 GB   disk2s1

```

Even though it's reported as an 'internal' drive, many other people may use this slot for regular SD cards. Sometimes I use an SD card in a USB adapter that's also formatted in HFS format, so the filesystem type isn't a good discriminator.

The output from system_report might be more useful. It shows that this drive has 'owner ship' enabled. (ignore ownership = no).Excluding drives with ownership enabled is quite like 95% accurate, but not foolproof.
Hence, a blacklist in the advanced config is probably the best solution.

```
$  system_profiler SPStorageDataType
Storage:

    Macintosh HD:

      Available: 34,28 GB (34.276.270.080 bytes)
      Capacity: 479,11 GB (479.110.004.736 bytes)
      Mount Point: /
      File System: APFS
      Writable: Yes
      Ignore Ownership: No
      BSD Name: disk1s1
      Volume UUID: 14820296-AF63-3B3D-91F5-551DC0FD5A0A
      Physical Drive:
          Device Name: OWC Aura
          Media Name: AppleAPFSMedia
          Medium Type: SSD
          Protocol: SATA
          Internal: Yes
          Partition Map Type: Unknown

    Transcend JetDrive:

      Available: 237,69 GB (237.691.359.232 bytes)
      Capacity: 259,68 GB (259.677.814.784 bytes)
      Mount Point: /Volumes/Transcend JetDrive
      File System: Journaled HFS+
      Writable: Yes
      Ignore Ownership: No
      BSD Name: disk2s1
      Volume UUID: 02AD767B-6949-33F0-B031-592788C0B5CF
      Physical Drive:
          Device Name: Built In SDXC Reader
          Media Name: Apple SDXC Reader Media
          Protocol: Secure Digital
          Internal: Yes
          Partition Map Type: MBR (Master Boot Record)

```

lurch, 2017-10-25:
The trouble with an explicit drive blacklist (I'm assuming you're referring to e.g. blacklisting `/dev/disk2` in your example above) is that e.g. on Linux, the drives get assigned device-names depending on the order they were detected. So if you have USB drives GreenDisk, BlueDisk and RedDisk, and then  plugged in BlueDisk first, then RedDisk, and finally GreenDisk, you might have BlueDisk as `/dev/sdb` (`/dev/sda` is usually the internal HDD), RedDisk as `/dev/sdc`, and GreenDisk as `/dev/sdd`; so if you wanted to explicitly blacklist GreenDisk, then you couldn't really blacklist `/dev/sdd` because the next time you boot up (or if you unplugged and replugged the drives in a slightly different order), GreenDisk might have a different device-name. I'm not sure if I've explained that very well, but hopefully you see what I mean.

I dunno about Mac, but I think Windows also auto-assigns drive letters (`D:`, `E:`, `F:`, etc.) based on the order that devices are plugged in?

mverleun, 2017-10-25:
The Mac is more or less the same. Filtering has to be done on the volume name and not the name of the physical device.
In macOS each volume get's a name, comparable to the labels in a Linux filesystem. The devices also have a UUID. Linux has a similar mechanism, but I don't know if Windows does.

One difference between the Jetdrive and other SD cards is that the drive is always present. It's used is if it were an internal disk. Even though this doesn't make the device name predictable, it's quite consistent during reboots if no other external drives are attached.

How about creating a (hidden) file detector? If a volume contains a specific hidden file, such as `.Exclude_from_Etcher` this could be detected and used to exclude the drive from the list?
Would this work for Windows as well?

lurch, 2017-10-25:
> it's quite consistent during reboots if no other external drives are attached.

The "if" in that sentence is exactly the sort of thing I'm worried about... ;-)
(it's far too easy for somebody to reboot with some other external drive still plugged in)

> How about creating a (hidden) file detector? If a volume contains a specific hidden file, such as .Exclude_from_Etcher this could be detected and used to exclude the drive from the list?

We _could_ look for specific files, but that then gets tricky as it assumes that the partition is already mounted before Etcher tries reading it, and a file-on-a-filesystem is a partition-level thing, whereas Etcher operates on a disk-level basis; so if there are multiple partitions on a disk Etcher would need to check each of them (which again, it can't do if the OS doesn't auto-mount all the partitions, e.g. a disk with ext4 partitions connected to a Windows PC).

I like your idea of using the UUID though, but as you say, any feature we add to Etcher ought to work the same across all three platforms.

**If** we add a blacklist feature, I'm thinking some kind of feature-combination might work? E.g. blacklist a disk if it has label XXXX **AND** size YYYY ? (**OR** uuid ZZZZ ?)  But that quickly gets complicated, which means it might fail, and we wouldn't want people to blame Etcher if they use it to overwrite a drive that they _thought_ they'd blacklisted.

Looking at the problem from a different angle, is there any other information we could display in the drive-selection window that would make it "easier" for you to avoid selecting the "wrong" drive?

jviotti, 2017-10-25:
All major operating systems have a concept similar to UUID to unique identify drives. In any case, we shouldn't make users manually pass UUIDs to the application. The drive selector modal should have a "blacklist" button alongside each of the drives that transparently fetches its unique id and saves it. Then, the user can potentially open a window from the settings page to see the blacklisted drives and whitelist any of those.

jhermsmeier, 2017-10-25:
Those UUIDs are usually per volume / mountable partition or the GPT's UUID. Both of which should be accessible to us:

- Windows tracks also this through `/System Volume Information/IndexerVolumeGuid` on the volume
- Mac OS has `/.fseventsd/.fseventsd-uuid` (also per volume afaik)
- Linux: not sure how, or if at all (?)

jviotti, 2017-10-25:
We could go with the serial number as well. If we add support for it on drivelist, then its a piece of cake.

mverleun, 2017-10-25:
There are indeed a lot of things to consider. Coming up with something that is 100% fault proof might be difficult. But combining different techniques could be feasible with a priority system?
If a UUID is available use it to blacklist a device. If not use serial number?

Linux supports UUID's on volume level and partition level, e.g.:
```
# ls -l /dev/disk/by-uuid/
total 0
lrwxrwxrwx 1 root root 10 okt  2 19:49 64124f0a-173b-49fc-a134-3898709a878a -> ../../sda3
lrwxrwxrwx 1 root root 10 okt  2 19:49 BF12-60C5 -> ../../sda1
lrwxrwxrwx 1 root root 10 okt  2 19:49 d5ba79d7-44e9-469d-bc25-57345c45477e -> ../../sda2

# ls -l /dev/disk/by-partuuid/
total 0
lrwxrwxrwx 1 root root 10 okt  2 19:49 011e5e86-102f-4990-a5bb-9059e39a0559 -> ../../sda2
lrwxrwxrwx 1 root root 10 okt  2 19:49 0a194633-0ae2-49dc-977a-338d768f3a28 -> ../../sda3
lrwxrwxrwx 1 root root 10 okt  2 19:49 7d3cc72e-d7aa-4d8d-8156-c062793c985a -> ../../sda1
```

And serial numbers are available  in Linux to:
```
# lsblk -o NAME,SERIAL  -a
\NAME   SERIAL
sda    5VJB25AF
├─sda1 
├─sda2 
└─sda3 
sdb    5VJB1Y5N
├─sdb1 
├─sdb2 
└─sdb3 
loop0  
loop1  
loop2  
loop3  
loop4  
loop5  
loop6  
loop7
```

or:
```
# ls -l /dev/disk/by-id/ | grep ata
total 0
lrwxrwxrwx 1 root root  9 okt  2 19:49 ata-ST9500420ASG_5VJB1Y5N -> ../../sdb
lrwxrwxrwx 1 root root 10 okt  2 19:49 ata-ST9500420ASG_5VJB1Y5N-part1 -> ../../sdb1
lrwxrwxrwx 1 root root 10 okt  2 19:49 ata-ST9500420ASG_5VJB1Y5N-part2 -> ../../sdb2
lrwxrwxrwx 1 root root 10 okt  2 19:49 ata-ST9500420ASG_5VJB1Y5N-part3 -> ../../sdb3
lrwxrwxrwx 1 root root  9 okt  2 19:49 ata-ST9500420ASG_5VJB25AF -> ../../sda
lrwxrwxrwx 1 root root 10 okt  2 19:49 ata-ST9500420ASG_5VJB25AF-part1 -> ../../sda1
lrwxrwxrwx 1 root root 10 okt  2 19:49 ata-ST9500420ASG_5VJB25AF-part2 -> ../../sda2
lrwxrwxrwx 1 root root 10 okt  2 19:49 ata-ST9500420ASG_5VJB25AF-part3 -> ../../sda3
```
Serialnumber detection seems to be possible on the Mac as well on device level.
```$ system_profiler SPCardReaderDataType
Card Reader:

    Built in SD Card Reader:

      Vendor ID: 0x14e4
      Device ID: 0x16bc
      Subsystem Vendor ID: 0x14e4
      Subsystem ID: 0x96bc
      Revision: 0x0010
      Link Width: x1
      Link Speed: 2.5 GT/s

        SDXC Card (Class 10):

          Product Name: J`JDL  
          Manufacturer ID: 0x74
          Revision: 2.0
          Serial Number: 637536167
          Manufacturing Date: 2016-06
          Specification Version: 3.0
          Capacity: 259.71 GB (259,710,255,104 bytes)
          Removable Media: Yes
          BSD Name: disk2
          Partition Map Type: MBR (Master Boot Record)
          Volumes:
            Transcend JetDrive:
              Available: 237.69 GB (237,691,351,040 bytes)
              Capacity: 259.68 GB (259,677,816,832 bytes)
              Writable: Yes
              File System: Journaled HFS+
              BSD Name: disk2s1
              Mount Point: /Volumes/Transcend JetDrive
              Content: Apple_HFS
              Volume UUID: 02AD767B-6949-33F0-B031-592788C0B5CF
```

I have no experience with Windows and agree that you don't want people to blame Etcher if they use it to overwrite a drive that they thought they'd blacklisted.

Would reversing the process help? All devices are blacklisted until a user whitelists one? That's more or less what users do in the current flow anyway.

Given the complexity I wouldn't be disappointed if you decide to leave things as they are. Upfront I didn't realise the complexity at all.

Thanks anyway for discussing this extensively.

lurch, 2017-10-25:
> Thanks anyway for discussing this extensively.

Thank you for your discussion too, and thanks for the research and info you've provided :+1: 

> Given the complexity I wouldn't be disappointed if you decide to leave things as they are. Upfront I didn't realise the complexity at all.

I suspect a lot of people don't realise how much hard work it takes, to make Etcher appear so "simple" ;-)

jviotti, 2017-12-04:
Should this be converted into a feature request ticket?

jhermsmeier, 2018-01-18:
Closing this in favor of https://github.com/resin-io/etcher/issues/1916
Dusplicate of #1916

jhermsmeier, 2018-01-18:
Duplicate of #1916

closed 2018-01-18 by jhermsmeier

OverWorldD, 2023-10-06:
I just did this... tgif bit me in the a**. Whats a good recorevery software? thiking ddrescue in linux