balena-io/etcher #1729: Etcher doesn't like this image, gives no feedback on why.

jaxxzer, 2017-09-15:
I have been preparing Raspberry Pi images in this manner (Ubuntu 16.04):

- SD card with expanded filesystem into the computer.
- Shrink EXT4 partition with GParted
- `parted <sd card device> -s print` to find End of ext4 partition in MB
- dd bs=1M count=#from parted+1
- right click resulting .img file in nautilus and click compress....zip

Now I have a .img.zip file, etcher has been perfectly happy with these *.img.zip files except for one.

[This](https://s3.amazonaws.com/downloads.bluerobotics.com/Pi/0.0.9/ardusub-raspbian-0.0.9.img.zip) is the .img.zip file in question. When I select this file from the file browse dialog, etcher behaves as if I had pressed 'cancel'. It returns to the main ui, it still says 'select image', and it gives no feedback on why the image I selected is not showing up. When I unzip this file, etcher lets me select and writes the resulting .img normally, and it functions normally in the Pi afterward. I have also tried compressing the image from the command line using `zip` with the same results.

[Here](https://s3.amazonaws.com/downloads.bluerobotics.com/Pi/0.0.10/ardusub-raspbian-0.0.10.img.zip) is another .img.zip prepared in the same manner, that works as expected.

jaxxzer, 2017-09-15:
PS: same behavior on Etcher 1.0.0 and 1.1.2.

jhermsmeier, 2017-09-15:
Looks like a potential bug in a dependency we use to handle zip files;

```
buffer.js:829 Uncaught RangeError: Index out of range
    at checkOffset (buffer.js:829)
    at Buffer.readUInt32LE (buffer.js:891)
    at FsRead.readUntilFoundCallback [as callback] (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/node-stream-zip/node_stream_zip.js:190)
    at FsRead.readCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/node-stream-zip/node_stream_zip.js:798)
    at FSReqWrap.wrapper [as oncomplete] (fs.js:603)
```

I'll do some fiddeling to see what's going on with that in detail.

jhermsmeier, 2017-09-15:
The zip file itself being faulty can also be ruled out:
```
$ unzip -v -t Downloads/ardusub-raspbian-0.0.9.img.zip
Archive:  Downloads/ardusub-raspbian-0.0.9.img.zip
    testing: ardusub-raspbian-0.0.9.img   OK
No errors detected in compressed data of Downloads/ardusub-raspbian-0.0.9.img.zip
```

jhermsmeier, 2017-09-15:
Turns out this was an upstream bug that has already been fixed in a newer version, so updating to node-stream-zip to 1.3.7 https://github.com/resin-io/etcher/pull/1730 will fix this.

lurch, 2017-09-19:
Reminds me of the "dinosaur error" we had with lzma a while back ;-)

jhermsmeier, 2017-09-22:
Closing as this has been fixed in https://github.com/resin-io/etcher/pull/1730 and will land in the next Etcher release.

closed 2017-09-22 by jhermsmeier

lurch, 2017-09-23:
@jaxxzer In the meantime, feel free to use https://resin-nightly-downloads.s3.amazonaws.com/etcher/2017-09-22/1.1.2%2Bcb7117a/etcher-1.1.2%2Bcb7117a-x86_64.AppImage