balena-io/etcher #1741: OSx .dmg Error: Invalid image

stdex, 2017-09-24:
- **Etcher version:**
1.1.2
- **Operating system and architecture:**
Ubutnu 16.04 x64
- **Image flashed:**
OS X El Capitan 10.11.6 
- **Do you see any meaningful error information in the DevTools?**
No

Try to select image InstallESD.dmg from OS X El Capitan 10.11.6
Get error: Error: Invalid image

[image]

jviotti, 2017-09-24:
Can you link us to that image? I don't know for sure, but I have the impression that is an installer for macOS but not an actual image you have to flash directly.

jhermsmeier, 2017-09-24:
@jviotti the installESD.dmg is the Mac OS installer image, that's all good – see https://github.com/resin-io/etcher/issues/1572 for example.

jhermsmeier, 2017-09-24:
Turns out that image is corrupted:
```
Error: Invalid footer signature: Expected 0x6b6f6c79, saw 0x9d72b920
```

And Mac OS seems to agree:
[image]

lurch, 2017-10-03:
@jhermsmeier If this is `resolution: wontfix`, should it be closed?

closed 2017-10-03 by jhermsmeier