balena-io/etcher #1749: Nothing happens

vnthn, 2017-10-03:
- **Etcher version: 1.12**
- **Operating system and architecture: Windows 10 Pro 64 Bit**
- **Image flashed: raspbian stretch lite**
- **Do you see any meaningful error information in the DevTools?    no**

DevTools shows the source code of the gui, but it looks fine.

Hi, I wanted to flash the image for the Raspberry Pi on its SD Card. I can successfully start the program Etcher and I can select the image which I want to flash. I select the image by clicking on 'Open' in the dialog. After I click on 'Open' nothing happens. The blue 'Select Image' Button turns dark blue as it has been pressed. When I move the Mouse back into the Etcher-Window it turns light blue as it was.
But even after waiting for 30min nothing happens. The program doesnt crash, I can reselect the image as often as I want, but still nothing happens after selection.

lurch, 2017-10-03:
That's **very** strange! :-/
Please can you paste the full contents of the DevTools console output? Even if you think it "looks fine" we might be able to spot something useful.
Also, could you try the "Portable" version of Etcher, to see if that behaves any differently?

Hmmm, something else you could try would be going into Etcher's settings (click on the cog icon) and enabling "Unsafe mode", in case Etcher is mis-identifying your SD card as a system drive.

vnthn, 2017-10-03:
Hi Andrew,

 

I fixed the problem by reformatting the SD Card with a partition tool. It held an old Raspberry Pi image, which was unreadable for Windows. I then successfully flashed the image via the Disk Imager.

I started Etcher again with no SD Card inserted and it successfully loads the image, then asking me where to push the image.

 

I don’t know where the problem is, but maybe Etcher could throw an error or at least a warning.

 

Although the problem is fixed, I have the output oft he console log attached.

 

Another problem is, that the application window turns complete white after I while. In the Task Manager I see there are four tasks. After the application turns white there is only one task running. Maybe that is a server crashing, which is responsible for rendering the application’s window contents? I cannot look at the DevTools by pressing Ctrl-Alt-I when the application is in this state.

 

Kind regards

Nils

lurch, 2017-10-05:
> I have the output oft he console log attached.

I'm guessing you must have sent it as an email attachment, and it looks like GitHub doesn't handle those? Please try replying using the Github web interface, and copy'n'pasting your console log as a comment. Thanks.

closed 2017-12-04 by jviotti