balena-io/etcher #1752: Blank Etcher window when installing latest version from redhat repo

sntyj, 2017-10-04:
- **Etcher version:**  1.1.2+3147a93-1
- **Operating system and architecture:** Fedora 26 64 bit (4.12.14-300.fc26.x86_64)
- **Image flashed:** From repo bintray--resin-io-redhat
- **Do you see any meaningful error information in the DevTools?**
N/A

On Fedora 26, running dnf upgrade -y installs version 1.1.2+3147a93-1. This version does not show any GUI elements. "Downgrading" and installing various previous versions allows normal usage of Etcher. It is only this latest version that brings up a blank window. Screenshot is attached.
[image]

Shou, 2017-10-05:
Are there any error messages in the dev tools (`Ctrl+Alt+i`)?

lurch, 2017-10-05:
Hmmm, that version (1.1.2+3147a93) is from our `etcher-devel` component, rather than our `etcher` component (which is still on 1.1.2), see https://bintray.com/resin-io/redhat

@sntyj Did you explicitly enable `etcher-devel`, or is there a possibility we've misconfigured our RPM repository and/or our [RPM instructions](https://github.com/resin-io/etcher#redhat-rhel-and-fedora-based-package-repository-gnulinux-x86x64)?
(BTW this is a side-question, and not _directly_ related to your initial bug report)

sntyj, 2017-10-05:
@Shou Sorry I had that shortcut mapped before so I couldn't get to the dev tools. I do have some errors. Please see below:

> /opt/Etcher/resources/app.asar/node_modules/resin-corvus/src/sentry.js:74 Uncaught TypeError: Cannot read property 'https' of undefined
    at Object.install (/opt/Etcher/resources/app.asar/node_modules/resin-corvus/src/sentry.js:74:40)
    at Object.keys.forEach (/opt/Etcher/resources/app.asar/node_modules/resin-corvus/src/resin-corvus.js:170:18)
    at Array.forEach (native)
    at Object.install (/opt/Etcher/resources/app.asar/node_modules/resin-corvus/src/resin-corvus.js:164:36)
    at Object.<anonymous> (/opt/Etcher/resources/app.asar/lib/gui/modules/analytics.js:24:13)
    at Object.<anonymous> (/opt/Etcher/resources/app.asar/lib/gui/modules/analytics.js:85:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at file:///opt/Etcher/resources/app.asar/lib/gui/app.js:42:19
/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4920 Uncaught Error: [$injector:modulerr] Failed to instantiate module Etcher due to:
Error: [$injector:nomod] Module 'Etcher' is not available! You either misspelled the module name or forgot to load it. If registering a module ensure that you specify the dependencies as the second argument.
http://errors.angularjs.org/1.6.3/$injector/nomod?p0=Etcher
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:66:12
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2257:17
    at ensure (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2178:38)
    at module (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2255:14)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4896:22
    at forEach (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:402:20)
    at loadModules (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4880:5)
    at createInjector (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4802:19)
    at doBootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1909:20)
    at bootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1930:12)
    at angularInit (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1815:5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:33340:5
    at HTMLDocument.trigger (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:3435:5)
http://errors.angularjs.org/1.6.3/$injector/modulerr?p0=Etcher&p1=Error%3A%…er%2Fresources%2Fapp.asar%2Fnode_modules%2Fangular%2Fangular.js%3A3435%3A5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:66:12
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2257:17
    at ensure (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2178:38)
    at module (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2255:14)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4896:22
    at forEach (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:402:20)
    at loadModules (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4880:5)
    at createInjector (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4802:19)
    at doBootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1909:20)
    at bootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1930:12)
    at angularInit (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1815:5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:33340:5
    at HTMLDocument.trigger (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:3435:5)
http://errors.angularjs.org/1.6.3/$injector/modulerr?p0=Etcher&p1=Error%3A%…er%2Fresources%2Fapp.asar%2Fnode_modules%2Fangular%2Fangular.js%3A3435%3A5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:66:12
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4920:15
    at forEach (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:402:20)
    at loadModules (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4880:5)
    at createInjector (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4802:19)
    at doBootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1909:20)
    at bootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1930:12)
    at angularInit (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1815:5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:33340:5
    at HTMLDocument.trigger (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:3435:5)

@lurch I did not enable etcher-devel through commands, although I wouldn't say that I'm not incapable of accidentally enabling something like that!

I did follow the RPM instructions you mentioned. I actually thought I misconfigured it at one point I did both methods using dnf and yum to install the package (after removing it each time). I added the repo using the following command (grabbed from shell history):

sudo wget https://bintray.com/resin-io/redhat/rpm -O /etc/yum.repos.d/bintray-resin-io-redhat.repo

SaharahSarah, 2017-10-09:
I'm having the same issue. I followed the [directions](https://github.com/resin-io/etcher#redhat-rhel-and-fedora-based-package-repository-gnulinux-x86x64) and the repo is set to https://resin-io.bintray.com/redhat/ which seems to include stable and development images.

sntyj, 2017-10-09:
Hello. New build came out today, and it seems that I'm still having the same problem with it. If I remove etcher-electron and reinstall it via dnf, it pulls version 1.1.2+179597b-1;

>[xxxx@XXXXXX ~]$ sudo dnf install etcher-electron
Last metadata expiration check: 0:04:17 ago on Mon 09 Oct 2017 09:32:21 AM CDT.
Dependencies resolved.
================================================================================
 Package           Arch     Version            Repository                  Size
================================================================================
Installing:
 etcher-electron   x86_64   1.1.2+179597b-1    bintray--resin-io-redhat    58 M

Side note, it is still loading as a blank window like before. Here is the error message from the debug console below:
>Uncaught TypeError: Cannot read property 'https' of undefined
    at Object.install (/opt/Etcher/resources/app.asar/node_modules/resin-corvus/src/sentry.js:74:40)
    at Object.keys.forEach (/opt/Etcher/resources/app.asar/node_modules/resin-corvus/src/resin-corvus.js:170:18)
    at Array.forEach (native)
    at Object.install (/opt/Etcher/resources/app.asar/node_modules/resin-corvus/src/resin-corvus.js:164:36)
    at Object.<anonymous> (/opt/Etcher/resources/app.asar/lib/gui/modules/analytics.js:24:13)
    at Object.<anonymous> (/opt/Etcher/resources/app.asar/lib/gui/modules/analytics.js:85:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at file:///opt/Etcher/resources/app.asar/lib/gui/app.js:42:19
/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4920 Uncaught Error: [$injector:modulerr] Failed to instantiate module Etcher due to:
Error: [$injector:nomod] Module 'Etcher' is not available! You either misspelled the module name or forgot to load it. If registering a module ensure that you specify the dependencies as the second argument.
http://errors.angularjs.org/1.6.3/$injector/nomod?p0=Etcher
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:66:12
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2257:17
    at ensure (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2178:38)
    at module (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2255:14)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4896:22
    at forEach (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:402:20)
    at loadModules (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4880:5)
    at createInjector (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4802:19)
    at doBootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1909:20)
    at bootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1930:12)
    at angularInit (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1815:5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:33340:5
    at HTMLDocument.trigger (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:3435:5)
http://errors.angularjs.org/1.6.3/$injector/modulerr?p0=Etcher&p1=Error%3A%…er%2Fresources%2Fapp.asar%2Fnode_modules%2Fangular%2Fangular.js%3A3435%3A5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:66:12
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2257:17
    at ensure (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2178:38)
    at module (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:2255:14)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4896:22
    at forEach (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:402:20)
    at loadModules (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4880:5)
    at createInjector (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4802:19)
    at doBootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1909:20)
    at bootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1930:12)
    at angularInit (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1815:5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:33340:5
    at HTMLDocument.trigger (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:3435:5)
http://errors.angularjs.org/1.6.3/$injector/modulerr?p0=Etcher&p1=Error%3A%…er%2Fresources%2Fapp.asar%2Fnode_modules%2Fangular%2Fangular.js%3A3435%3A5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:66:12
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4920:15
    at forEach (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:402:20)
    at loadModules (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4880:5)
    at createInjector (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:4802:19)
    at doBootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1909:20)
    at bootstrap (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1930:12)
    at angularInit (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:1815:5)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:33340:5
    at HTMLDocument.trigger

lurch, 2017-10-25:
We've now fixed our snapshot builds :tada: ( #1795 )
Can the people who were affected by this "blank window" issue please try updating their etcher-electron package and let us know if it's all working now?

The issue with snapshot releases getting installed on Fedora in the first place should hopefully be fixed soon ( #1767 ).

sntyj, 2017-10-26:
As of today, ran dnf upgrade -y and now I have a working version of etcher. Thank you!
[image]

closed 2017-10-27 by lurch