balena-io/etcher #1754: .zip: cannot see img file with accentuated characters

manu-t, 2017-10-06:
- **Etcher version:** 1.1.12
- **Operating system and architecture:** Win7 64 bits
- **Image flashed:** rpi_robotisé.zip
- **Do you see any meaningful error information in the DevTools?**
The decompressed file rpi_robotisé.img is fine, but the zip file doesn'work.
I presume accentuated charaters is the culprit.
if a "rpi_robitisé.img" file is in a test.zip file, it doesn'work either.
Maybe related to #1394

lurch, 2017-10-12:
Hmm, I thought we'd fixed all our filename-encoding problems, but looks like you've found a new one ;-)

Please can you provide a link to the `rpi_robotisé.zip` so that we can download and test it ourselves?

manu-t, 2017-10-12:
Well, you can rename any .zip file to "rpi_robotisé.zip" for example, it should reproduce the bug.
If it's not enough and you need a real image, I can send you a private link as it's kinda professionnal... so I don't know if I can share it.

lurch, 2017-10-12:
Hmm, I just did some testing using a dummy image file I made, and in all combinations I tried it works okay :-/

| Zip filename | Image filename | Works in Etcher 1.1.2 |
| --- | --- | --- |
| _n/a_ | `rpi_robitisé.img` | :heavy_check_mark: |
| _n/a_ | `test.img` | :heavy_check_mark: |
| `rpi_robitisé1.zip` | `rpi_robitisé.img` | :heavy_check_mark: |
| `rpi_robitisé2.zip` | `test.img` | :heavy_check_mark: |
| `test1.zip` | `rpi_robitisé.img` | :heavy_check_mark: |
| `test2.zip` | `test.img` | :heavy_check_mark: |
| `rpi_robitisé.zip` | `rpi_robitisé.img` | :heavy_check_mark: |

Perhaps there's something 'special' about how your zipfile is created? Please try testing with https://resin-nightly-downloads.s3.amazonaws.com/etcher/2017-09-22/1.1.2%2Bcb7117a/Etcher-Setup-1.1.2%2Bcb7117a-x64.exe which uses a newer version of `node-stream-zip`, and if that doesn't work then please send a private link to `hello@etcher.io`

manu-t, 2017-10-12:
I sent you a link to the image in your e-mail. (The zip file is generated by 7-zip)
I didn't manage to make the new executable work (missing dll or other ressources as there are no images on the screen).

lurch, 2017-10-13:
I've received your email and downloaded your file, thank you. And I can confirm that Etcher is unable to write the image - after selecting the zipfile, Etcher behaves as though no file was selected at all :-/

I wonder if it might be some kind of filename encoding-issue within the zipfile itself though? I'll do a more thorough investigation later, but I noticed that (on Linux) `unzip -l ~/Downloads/rpi_robotisé.zip` lists the enclosed filename as `rpi_robotisВ.img` whereas `7z l ~/Downloads/rpi_robotisé.zip` displays `rpi_robotis.img`. What version of 7-zip are you creating this zipfile with?
For comparison, both `unzip` and `7z` correctly display the enclosed filename in my test `rpi_robitisé1.zip` as `rpi_robitisé.img`.  (Haha, I've just realised I named my test files with `..robiti..` instead of `..roboti..`)

manu-t, 2017-10-13:
I have a pretty old 7-zip : v9.20. I will try to update it...

manu-t, 2017-10-14:
same with 7-zip 16.04 64 bits

lurch, 2017-10-14:
Cool, thanks for trying with the updated version. We'll try to dig into this problem soon.

jhermsmeier, 2017-11-14:
@lurch where is that .zip file? I'd like to have a look to see what's wrong there.

lurch, 2017-11-14:
Argh, I totally forgot about this, sorry @manu-t !

@jhermsmeier I'll send you a PM...

closed 2018-10-15 by jviotti