balena-io/etcher #1756: Warnings specifically for NOOBS RaspberryPi "image"

paul-hammant, 2017-10-08:
- **Etcher version:**

1.1.2

- **Operating system and architecture:**

2017 Mac, latest OS.

- **Image flashed:**

NOOBS.zip from RaspberryPi Foundation: https://downloads.raspberrypi.org/NOOBS_latest

- **Suggestion**

Warning dialog that end user has not actually selected an image (most on the RaspberryPi site are images), but just a zip. Then a suggestion that the end user could go back to the Pi site and read more details.

#1162 relates, but **this** is a different ask.

lurch, 2017-10-12:
@paul-hammant I've left a reply to your comment in the other issue.

> Then a suggestion that the end user could go back to the Pi site and read more details.

Obviously Etcher gets used to write disk images for other platforms than the Raspberry Pi, so if we were to offer any advice like you're suggesting, then we'd have to be **sure** that the user was trying to flash a NOOBS zipfile for Raspberry Pi, and not accidentally trying to flash some other zipfile for a different platform. There's currently no attempt in Etcher to "identify" any particular images / zipfiles, and again doing so would be a non-trivial task (unless we went for something as crude as trying to match just the filename).

The [NOOBS download page](https://www.raspberrypi.org/downloads/noobs/) has links to a number of setup guides. However I've just noticed that none of those guides mentions the issue with SD Formatter and 64GB SDXC cards, so I've just added an issue for that https://github.com/raspberrypilearning/software-guide/issues/13

paul-hammant, 2017-10-12:
That's my point, and I got flicked off in GitHub issues in the pertinent Raspberry-Pi repo. Yay insta-closing of issues (not).

lurch, 2017-10-12:
I also just created this PR: https://github.com/raspberrypi/documentation/pull/744

paul-hammant, 2017-10-12:
Good work.

lurch, 2017-10-17:
@paul-hammant FYI https://github.com/raspberrypilearning/software-guide/issues/13 and https://github.com/raspberrypi/documentation/pull/744 have now both been fixed :-)

paul-hammant, 2017-10-17:
:)

closed 2017-10-17 by paul-hammant