balena-io/etcher #1765: can't run Etcher on Fedora 27

andreasn, 2017-10-11:
Downloaded Etcher from the website, since that's the tool that the retropie website recommended. Tried to run it, both from the desktop and via the terminal without any luck.

- **Etcher version:**
1.1.2

- **Operating system and architecture:**
Fedora Linux 27, x86

- **Image flashed:**
none, didn't get that far

- **Do you see any meaningful error information in the DevTools?**
This is the output from the cli when I ran `./etcher-1.1.2-x86_64.AppImage `
`/tmp/.mount_hC3ysX/usr/bin/etcher-electron: error while loading shared libraries: libXss.so.1: cannot open shared object file: No such file or directory`

Shou, 2017-10-12:
Either installing the Etcher `rpm` from the website as opposed to the AppImage or installing `libXScrnSaver` through your package manager should resolve this (`sudo dnf install libXScrnSaver`?). Let me know how it goes!

lurch, 2017-10-12:
Closing as duplicate of #835 
See also #1005 and https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#runtime-gnulinux-dependencies

As @Shou says, if you use https://github.com/resin-io/etcher#redhat-rhel-and-fedora-based-package-repository-gnulinux-x86x64 then `libXScrnSaver` should be installed automatically when you install `etcher-electron` (although I'm afraid there's currently a problem with our RPM repo setup #1767 which means people using our RPM repo are getting broken versions of Etcher #1752 :slightly_frowning_face: ).

closed 2017-10-12 by lurch