balena-io/etcher #1766: can i uninstall i need answers fast HELP!

Whoeverhitme, 2017-10-11:
- **Etcher version:** ???
- **Operating system and architecture:**
- **Image flashed:**  retro pie
- **Do you see any meaningful error information in the DevTools?**
it wont let me uninstall because there is no uninstall

jviotti, 2017-10-12:
Its hard to offer you help if we don't know the operating system you're running. In can any case, Etcher can be uninstalled with the standard uninstall method of your OS of choice.

Whoeverhitme, 2017-10-12:
thank you 
I was putting in a retro pie on my rpi 3 
and I couldn't fix it so I formated it because it was stuck in a boot loop.
just for future reference do you know where the uninstall button is 
I don't know

Whoeverhitme, 2017-10-12:
and also when I download etcher it works then If I close it I searched my whole computer and I cant find etcher

jviotti, 2017-10-13:
I'm sorry but I can't help if without knowing the operating system that your laptop/computer where you have Etcher is running. In any case, the uninstall method is the standard one from your OS. i.e. if you have Windows, then you can uninstall Etcher from the Apps menu in the Settings application.

jhermsmeier, 2017-10-13:
@Whoeverhitme are you asking about how to uninstall Etcher, or how to uninstall what you flashed onto the SD-card?

jhermsmeier, 2017-10-24:
Closing this due to lack of response; @Whoeverhitme feel free to re-open if you come back to this

closed 2017-10-24 by jhermsmeier