balena-io/etcher #1769: Do not put AppImages into archives

TheAssassin, 2017-10-13:
Currently the AppImages (which are themselves compressed files) are put into `.zip` archives. Please consider uploading them as-is. Here's why:

 * Putting them into archives adds nothing (especially) compression wise, and wastes computer runtime (which is against the principles of Green IT, and is super annoying anyway)
 * Archives add an additional step, the user has to extract them. This ruins the selling argument for AppImages: the 3 step workflow -- download, make executable, run.
 * The additional step of extraction also prevents two of the most useful tools from working:
   * appimaged, an optional user space daemon integrating freshly downloaded AppImages with the Desktop environment, cannot detect AppImages in archives. Thus, Etcher is not recognized, and can't be put into the launcher etc. automatically
   * AppImageUpdate cannot work with archives of any kind. This super easy-to-use method, which has support for GitHub releases, makes it super easy to update the AppImages. It works with a binary delta sync algorithm, and is quite efficient. Also, AppImageUpdate integrates into appimaged, which allows the user to right click the application icon in the launcher and hit "Update".

It should be fairly easy to get rid of the archiving. And if you consider implementing AppImageUpdate, that'd be awesome.

CC #1476, relates to AppImageUpdate functionality.

lurch, 2017-10-14:
See #772 for the original reason why we did this.
However, you'll be pleased to hear that this has already been fixed by #1703 :grinning:  (but there's not yet been a production release of Etcher made since that change)

closed 2017-10-14 by lurch

TheAssassin, 2017-10-14:
Sorry, I did perform a few searches, but wasn't able to dig out those two for whyever reason. Great to hear it's fixed.